mpromonet/webrtc-streamer #589: error: cannot use ‘typeid’ with ‘-fno-rtti’

iqbal-h, 2023-05-11:
**Describe the bug and Steps to reproduce**
I want to add some code in desktopcapturer.cc that requires Torch, TorchVision, and OpenCV. 
So I added the necessary header.h and code.cc file in webrtc-streamer/inc and ../src
Then I modified the CMakeList.txt. After the following existing code of the CMakeList.txt:
```
# webrtc 
set (WEBRTCINCLUDE ${WEBRTCROOT}/src ${WEBRTCROOT}/src/third_party/abseil-cpp ${WEBRTCROOT}/src/third_party/jsoncpp/source/include  ${WEBRTCROOT}/src/third_party/jsoncpp/generated ${WEBRTCROOT}/src/third_party/libyuv/include)
target_include_directories(${CMAKE_PROJECT_NAME} PRIVATE ${WEBRTCINCLUDE})
FILE(GLOB_RECURSE LIBWEBRTC ${WEBRTCOBJS}/${CMAKE_STATIC_LIBRARY_PREFIX}*${CMAKE_STATIC_LIBRARY_SUFFIX})
target_link_libraries (${CMAKE_PROJECT_NAME} ${WEBRTCOBJS}/rtc_base/${CMAKE_STATIC_LIBRARY_PREFIX}rtc_json${CMAKE_STATIC_LIBRARY_SUFFIX} ${LIBWEBRTC}) 
add_definitions(-DHAVE_JPEG)
```

I added the following lines:
```
# find Torch, OpenCV, and TorchVision
find_package(Torch REQUIRED)
find_package(OpenCV REQUIRED)
find_package(TorchVision REQUIRED)

# include their directories
target_include_directories(${CMAKE_PROJECT_NAME} PRIVATE ${TORCH_INCLUDE_DIRS} ${OpenCV_INCLUDE_DIRS} ${TorchVision_INCLUDE_DIRS})

# link them to your target
target_link_libraries(${CMAKE_PROJECT_NAME} ${TORCH_LIBRARIES} ${OpenCV_LIBS} ${TorchVision_LIBRARIES})
```
- Then Run `cmake -DCMAKE_PREFIX_PATH=/home/hiqbal/libtorch .`
- Then Run `make`

The following errors occur:
[image]
and
[image]

It seems that OpenCV and Torch were not happy with the presence of `-fno-rtti `flag in the `CMakeList.txt` in the following two places:
```
elseif (APPLE)
...
    add_definitions(-DWEBRTC_MAC -DWEBRTC_POSIX -fno-rtti)
   ...
else()
    # libv4l2cpp
    ...
    # webrtc
    add_definitions(-DWEBRTC_POSIX -fno-rtti)
    ...
```
So I removed this flag from the add_definitions and ran `make`. There previous errors were sorted but now I have the following error:

[image]

I understand this problem is not directly related to webrtc-streamer. However, most expertise on the subject exists with the contributors in this repo.
I would really appreciate the help.
Thank you.

**Expected behavior**
Make should successfully create ./webrtc-streamer executable

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Browser Chrome

mpromonet, 2023-05-11:
Hi,

You might try to run `gn gen` with `use_rtti=true` and remove the ` -fno-rtti`

Best Regards,
Michel.

iqbal-h, 2023-05-11:
Hi,

Thanks. To try this clean. I removed my code and additions in CMakeList.txt and restored webrtc-streamer as it is. 
Then I removed the `-fno-rtti` from the following:
```
elseif (APPLE)
...
    #add_definitions(-DWEBRTC_MAC -DWEBRTC_POSIX -fno-rtti)
   add_definitions(-DWEBRTC_MAC -DWEBRTC_POSIX)
   ...
else()
    # libv4l2cpp
    ...
    # webrtc
    #add_definitions(-DWEBRTC_POSIX -fno-rtti)
   add_definitions(-DWEBRTC_POSIX)
    ...
```

and edited line 118 of CMakeLists.txt as follows:

`EXECUTE_PROCESS(WORKING_DIRECTORY ${WEBRTCROOT}/src/out/${CMAKE_BUILD_TYPE} COMMAND ${SHELLCOMMAND} gn gen . --args=use_rtti=true)  `

`cmake .` worked fine.
`make` reached 100% but then generated following errors. I am just pasting few of them below for reference.

Any thoughts?

Another solution I am thinking is to make a static library of the Torch and Open CV code say `myLib.a` and then add this library to `CMakeLists.txt` and simply do the myLib::function calls in `desktopcapturer.cpp`. Will this work or building even this result in similar issues?

```
/usr/bin/ld: CMakeFiles/webrtc-streamer.dir/src/HttpServerRequestHandler.cpp.o: in function `RequestHandler::handle(CivetServer*, mg_connection*)':
HttpServerRequestHandler.cpp:(.text._ZN14RequestHandler6handleEP11CivetServerP13mg_connection[_ZN14RequestHandler6handleEP11CivetServerP13mg_connection]+0x2f8): undefined reference to `Json::Value::asString[abi:cxx11]() const'
/usr/bin/ld: HttpServerRequestHandler.cpp:(.text._ZN14RequestHandler6handleEP11CivetServerP13mg_connection[_ZN14RequestHandler6handleEP11CivetServerP13mg_connection]+0x4c4): undefined reference to `Json::writeString[abi:cxx11](Json::StreamWriter::Factory const&, Json::Value const&)'
/usr/bin/ld: HttpServerRequestHandler.cpp:(.text._ZN14RequestHandler6handleEP11CivetServerP13mg_connection[_ZN14RequestHandler6handleEP11CivetServerP13mg_connection]+0x620): undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: CMakeFiles/webrtc-streamer.dir/src/HttpServerRequestHandler.cpp.o: in function `WebsocketHandler::handleData(CivetServer*, mg_connection*, int, char*, unsigned long)':
HttpServerRequestHandler.cpp:(.text._ZN16WebsocketHandler10handleDataEP11CivetServerP13mg_connectioniPcm[_ZN16WebsocketHandler10handleDataEP11CivetServerP13mg_connectioniPcm]+0x133): undefined reference to `Json::writeString[abi:cxx11](Json::StreamWriter::Factory const&, Json::Value const&)'
/usr/bin/ld: HttpServerRequestHandler.cpp:(.text._ZN16WebsocketHandler10handleDataEP11CivetServerP13mg_connectioniPcm[_ZN16WebsocketHandler10handleDataEP11CivetServerP13mg_connectioniPcm]+0x201): undefined reference to `Json::Value::asString[abi:cxx11]() const'
/usr/bin/ld: HttpServerRequestHandler.cpp:(.text._ZN16WebsocketHandler10handleDataEP11CivetServerP13mg_connectioniPcm[_ZN16WebsocketHandler10handleDataEP11CivetServerP13mg_connectioniPcm]+0x419): undefined reference to `Json::writeString[abi:cxx11](Json::StreamWriter::Factory const&, Json::Value const&)'
/usr/bin/ld: CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o: in function `PeerConnectionManager::getStreamList()':
PeerConnectionManager.cpp:(.text+0x1fb5): undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o: in function `PeerConnectionManager::PeerConnectionObserver::OnIceCandidate(webrtc::IceCandidateInterface const*)':
PeerConnectionManager.cpp:(.text+0x2465): undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: PeerConnectionManager.cpp:(.text+0x24ee): undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o: in function `PeerConnectionManager::getPeerConnectionList()':
PeerConnectionManager.cpp:(.text+0x2d63): undefined reference to `Json::Value::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: PeerConnectionManager.cpp:(.text+0x2ed4): undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: PeerConnectionManager.cpp:(.text+0x2f6c): undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: PeerConnectionManager.cpp:(.text+0x30e4): undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: PeerConnectionManager.cpp:(.text+0x317b): undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: PeerConnectionManager.cpp:(.text+0x3293): undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: PeerConnectionManager.cpp:(.text+0x33a6): undefined reference to `Json::Value::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
[... cut ...]
/usr/bin/ld: ../webrtc/src/out/Release/obj/common_video/common_video/video_frame_buffer_pool.o:./../../buildtools/third_party/libc++/trunk/include/list:1718: more undefined references to `std::Cr::__libcpp_verbose_abort(char const*, ...)' follow
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/webrtc-streamer.dir/build.make:1133: webrtc-streamer] Error 1
make[1]: *** [CMakeFiles/Makefile2:124: CMakeFiles/webrtc-streamer.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

closed 2023-12-30 by mpromonet