mpromonet/webrtc-streamer #591: 添加-c参数后，仍然报错net::ERR_SSL_PROTOCOL_ERROR

1s1u, 2023-05-19:
我服务器docker安装镜像之后，通过命令行启动webRtc-streamer，但仍然报错net::ERR_SSL_PROTOCOL_ERROR，证书不是有两个参数么，为什么-c后面只有一个变量。
        也有可能是我-c后面的参数出现了问题，还想请教一下，-c具体怎么带参数

1s1u, 2023-05-19:
how tov write the correct format for -c
[image]

yangjieshao, 2023-06-20:
https://github.com/mpromonet/webrtc-streamer/issues/505

> 如果证书参数的格式正确的话，需要安装和配置openSSL Except for correct format of certificate, you also need to install and config openSSL

yangjieshao, 2023-06-20:
别折腾了 用Nginx转发一下得了

mpromonet, 2023-07-08:
對不起，我的中文很有限

closed 2023-07-08 by mpromonet

blwinner, 2024-02-27:
> 對不起，我的中文很有限
 
thanks for your amizing work.  he start webrtc-streamer by docker, and get error code: net::ERR_SSL_PROTOCOL_ERROR, so his question is : argument '-c' only one input, and create credential need two (i dont understant too:)), how can he start with '-c' in detail

mpromonet, 2024-02-27:
Hi @blwinner,

Did you look to https://github.com/mpromonet/webrtc-streamer/discussions/592 ?

Best Regards,
Michel.

tangxinqi0218, 2024-05-29:
> 别折腾了 用Nginx转发一下得了

是的，配置转发分分钟就搞定