mpromonet/webrtc-streamer #595: the video stuck

Cmy-art, 2023-05-31:
**Hi,Firstly I have to thank you for your open source,It's an awesome tool. 
But recently,when I use the webrtc-streamer to play a certain manufacturer's camera,it works servral seconds,and soon stuck
The following is the log, could you help analyze the reason? It would be great if you could tell me the solution,
Thank you very much and have a nice day!**
[log.txt](https://github.com/mpromonet/webrtc-streamer/files/11607949/log.txt)

Cmy-art, 2023-05-31:
I noticed the fragment "Same/old NTP timestamp (1682463356176 <= 1685498991970) for incoming frame. Dropping."
Is it due to the camera needing a calibration timestamp? @mpromonet

mpromonet, 2023-05-31:
Hi,
It may be camera sending b frame. You may try using option -o that doesnot uncompress or disable b frame in camera configuration. Or modifify the webrtc video decoding.
Best Regards
Michel

Cmy-art, 2023-05-31:
I used the option -o from the beginning,but it doesn't work.
On the contrary, I didn't use the -o parameter and the console only printed some error messages, but it still played normally.
In addition, I calibrated the camera clock with the option -o and the situation improved.
These phenomena make me a little confused, so I need your help @mpromonet