mpromonet/webrtc-streamer #597: oniceconnectionstatechange  state: disconnected

Yanxiaolong1998, 2023-06-01:
[image]
[image]

I use it with cloud server, but console log reminder me oniceconnectionstatechange  is disconnected, but server log is not close session.

Yanxiaolong1998, 2023-06-01:
[image]
I find if server have a lot of network card , web page  can't get flow, because i user it with other server, the status is good!

hack-fang, 2023-07-13:
try this [turnserver](https://juejin.cn/post/7030316400354803726) ,it help me to fix this problem.

closed 2025-04-26 by mpromonet