mpromonet/webrtc-streamer #603: How to get correct m_videoaudiomap for v4l2 devices when two devices with the same 'PRODUCT' plugged in?

rouxianmantou, 2023-07-05:
**Describe the bug**
1st: I use Ubuntu with two HDMI-USB capture card(v4l2 device, same brand and model) plugged in.
2nd: When connecting to the stream, the video and audio may not match.

**To Reproduce**
same with [Describe the bug]

**What happens**
Firstly, one Linux with two HDMI-USB capture card(v4l2 device, same brand and model) plugged in.
From the code(as shown in the Screenshots), I found in PeerConnectionManager.cpp#m_videoaudiomap stores the map of video and audio of v4l2 devices.

Then in the function 'V4l2AlsaMap.h#std::map<std::string,std::string>  getV4l2AlsaMap()', it exec 'getVideoDevices()' and 'getAudioDevices()', we can get 'VideoDeviceMap<videoDevName, devId>' and 'AudioDeviceMap<devId, audioDevName>'.
And then, in 'V4l2AlsaMap.h#std::string getDeviceId(const std::string& evt)', the video and audio is group by 'PRODUCT' or 'PCI_SUBSYS_ID' in 'device/uevent'.
But when two devices has same 'PRODUCT', the video and audio may not match.

I don't know match about Linux. Is there any way to avoid the situation, maybe use other info to group video and audio?

**Screenshots**
[image]
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Version: latest(07.04)

rouxianmantou, 2023-07-05:
Hello mpromonet,

I  plan to use the realpath of the symlink of device(/sys/class/video4linux(sound)/device), it can be x-y:m.n(video: 3-1:1.0 audio: 3-1:1.2, usb port), add x-y to the deviceId, I think the video and audio should match when x-y(both 3-1) is the same.
emmm I will do this for now, hope no bugs.

If you have some idea, pls help to tell me, thanks a lot!

rouxianmantou, 2023-07-06:
Hello @mpromonet 
another related question QAQ
the m_videoaudiomap is from devices in /sys/class/video4linux(sound)
when call /api/getMediaList api, it will find audio device with the devname from webrtc api, it could not match, I think?

mpromonet, 2023-07-08:
Hi @rouxianmantou 

You are right, in the past device name was used, but as it is not unique, I change my mind and use device number.
This part of the code need to be adapt to the new behavior.

Best Regards,
Michel.

closed 2023-07-08 by mpromonet

rouxianmantou, 2023-07-10:
Thanks a lot!

commit 4ab32d64 (2023-07-08): fix #603: fix lin nk detection between video and audio devices