mpromonet/webrtc-streamer #606: When I requested an rtsp stream, I found that webrtc requested it twice. What is the reason for this?

liusuyi2021, 2023-07-14:
**Describe the bug**
When I requested an rtsp stream, I found that webrtc requested it twice. What is the reason for this?
[image]

But when I used the configuration file to preview rtsp, I requested again it is once.
[image]
[image]

please tell me why????

mpromonet, 2023-07-14:
Hi,

Once for video, once for audio.

Best Regards,
Michel.

liusuyi2021, 2023-07-14:
how use once for video and not use audio？
怎样只保留一次rtsp请求

mpromonet, 2023-07-14:
Hi,

You need to modify source code, for now if audio is not set it use same url than video.

Best Regards,
Michel.

liusuyi2021, 2023-09-05:
最新版本0.8.2 每次请求4次rtsp流

mpromonet, 2023-09-05:
Hi @liusuyi2021 

Probably you are using different arguments like tcp/udp, width, ..., if stream cannot by shared, it openned new one. You could see this using page admin.html.

Best Regards,
Michel.