mpromonet/webrtc-streamer #607: 配置华为摄像机系统直接奔溃

handsometong, 2023-07-18:
配置华为的摄像头无法播放摄像头且摄像机直接奔溃，海康的摄像头可以使用
[image]

mpromonet, 2023-07-23:
Hi @handsometong 

Using a debugger will help you to find the cause of exception that is raised.

Best Regards,
Michel.

handsometong, 2023-07-24:
The Huawei camera system crashes directly。Program log message see below screemshot.
[image]

mpromonet, 2023-07-24:
Hi @handsometong,

Then parsing string is too weak for your camera, a debugger will help you to localize your problem. Increasing log may also help you.

Best Regards,
Michel.

handsometong, 2023-07-25:
I call api（ /api/log?level=0） and got new program log message below screemshot.How do I fix this bug and can you help me？

[image]

mpromonet, 2023-07-29:
Hi @handsometong 

I guess running in a debugger like gdb will help you.

Best Regards,
Michel.

flowersBloom826, 2025-04-27:
> 华为的摄像头配置多slice编码，无法播放监控流会导致系统直接闪退，关闭多slice编码可以正常使用 <img alt="image" width="1030" src="https://private-user-images.githubusercontent.com/12708876/254199839-36add542-2ead-451f-b599-20b94dc0c610.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU3MzU5MzAsIm5iZiI6MTc0NTczNTYzMCwicGF0aCI6Ii8xMjcwODg3Ni8yNTQxOTk4MzktMzZhZGQ1NDItMmVhZC00NTFmLWI1OTktMjBiOTRkYzBjNjEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDI3VDA2MzM1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY5ZGExOTJiNDJhMDAyOWFhZWYzMjlkNzliOWFjMWM4Zjg5NWQ3YjM5YWZmZGJmOTY2OGZhMTA2MDUzZmI3ZGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9Ec8T6P4q5KruJ7RISrNILuZELl8uhexFRluVtrokd4"> <img alt="image" width="1272" src="https://private-user-images.githubusercontent.com/12708876/254221082-a5c7cabf-cf54-43ad-a8e2-b5f6fe4e394d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU3MzU5MzAsIm5iZiI6MTc0NTczNTYzMCwicGF0aCI6Ii8xMjcwODg3Ni8yNTQyMjEwODItYTVjN2NhYmYtY2Y1NC00M2FkLWE4ZTItYjVmNmZlNGUzOTRkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDI3VDA2MzM1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0MjViMWYxZjhjMWU0ZTRhYjk0NDZmNGUwZDIxYThjZmZhMTQzYjAyNzZkYmZhNzkwNjM5MDQyNzdjOWNjNGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nITA5Tu6P8LZwoo_Aoy8VQwXaU2bnqdTjHFm8z-As1c">

ps（楼主，你这个华为摄像头的加上 -o后是否由花屏的现象，如果没有的话后台配置是怎么的呢，望回复，感谢）
 Hey OP, have you noticed screen weirdness with your Huawei camera after adding the "-o"? If not, what's the backend setup you've got going on there? Would love to, thanks!

commit 1018b11a (2023-08-06): this may fix #607 fixing nalu size when offset is not 0
closed 2025-05-17 by mpromonet