mpromonet/webrtc-streamer #611: H265 RTSP need support

dinggf123, 2023-08-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mpromonet, 2023-08-01:
Hi @dinggf123 

Feel free to submit a PR to add this support.

Best Regards,
Michel.

liusuyi2021, 2023-08-14:
我已经找到一个临时解决265的rtsp的播放方案，就是使用ffmpeg配合mediamtx进行转码 目前试用了几路视频 感觉还可以，你也可以试试。

closed 2023-09-03 by mpromonet

liusuyi2021, 2023-09-05:
谷歌已经支持HEVC的H265了，使用最新版本0.8.2 还是无法播放呢？