mpromonet/webrtc-streamer #614: Run the docker on jetson nano

RezaFeng, 2023-08-10:
I just run the docker image on jetson nano and try to expose the V4L2 device，but could get the image from the website. I am sure the usb camera is fine。
'''
docker run --device=/dev/video0 -p 8000:8000 -it mpromonet/webrtc-streamer
'''
I use this example to test it.

mpromonet, 2023-08-10:
Hi,

Looking to the logs may help you.

Best Regards
Michel

RezaFeng, 2023-08-10:
This is my logs, I can not find the v4l2 url
`Version:v0.8.1-18-g6b267af/Linux-arm64 civetweb@v1.16 webrtc@c71bfcc-dirty live555helper@45cbd98
{
	"urls" : 
	{
		"Allendale" : 
		{
			"position" : "42.972250,-85.953650",
			"video" : "rtsp://71.83.5.156/axis-media/media.amp"
		},
		"Bedford Hills" : 
		{
			"video" : "rtsp://73.114.177.111/axis-media/media.amp"
		},
		"BigSky" : 
		{
			"position" : "45.284650,-111.368290",
			"video" : "rtsp://76.75.8.120/axis-media/media.amp"
		},
		"Great Falls" : 
		{
			"position" : "47.551039,-111.539400",
			"video" : "rtsp://76.75.8.116/axis-media/media.amp"
		},
		"Honk-Kong" : 
		{
			"options" : "rtptransport=tcp&timeout=60",
			"position" : "22.352734,114.1277",
			"video" : "rtsp://weathercam.gsis.edu.hk/axis-media/media.amp"
		},
		"Mendoza" : 
		{
			"position" : "-32.890840,-68.827170",
			"video" : "rtsp://190.113.172.8/"
		},
		"Montana" : 
		{
			"video" : "rtsp://64.187.201.16/axis-media/media.amp"
		},
		"Nordland" : 
		{
			"position" : "67.280000,14.405010",
			"video" : "rtsp://77.110.228.219/axis-media/media.amp"
		},
		"Norwich" : 
		{
			"video" : "rtsp://37.157.51.30/axis-media/media.amp"
		},
		"Orlando" : 
		{
			"position" : "28.538340,-81.379240",
			"video" : "rtsp://97.68.104.34/axis-media/media.amp"
		},
		"Ottawa" : 
		{
			"position" : "45.402420,-75.653957",
			"video" : "rtsp://67.69.107.188:554/axis-media/media.amp"
		},
		"Pocapaglia" : 
		{
			"video" : "rtsp://176.65.94.105/axis-media/media.amp"
		},
		"PriceCenterPlaza" : 
		{
			"video" : "rtsp://132.239.12.145:554/axis-media/media.amp"
		},
		"Reykjavik" : 
		{
			"position" : "64.135480,-21.895410",
			"video" : "rtsp://157.157.138.235/axis-media/media.amp"
		},
		"Test" : 
		{
			"video" : "rtsp://demo:demo@ipvmdemo.dyndns.org:5541/onvif-media/media.amp?sessiontimeout=60&streamtype=unicast"
		},
		"Trento" : 
		{
			"video" : "rtsp://194.32.173.211/axis-media/media.amp"
		},
		"Vaison-La-Romaine" : 
		{
			"position" : "44.238960,5.074610",
			"video" : "rtsp://176.139.87.16/axis-media/media.amp"
		},
		"Vancouver" : 
		{
			"position" : "49.249660,-123.119340",
			"video" : "rtsp://174.6.126.86/axis-media/media.amp"
		},
		"VyhladJazero" : 
		{
			"video" : "rtsp://stream.strba.sk:1935/strba/VYHLAD_JAZERO.stream"
		},
		"Wershofen" : 
		{
			"position" : "50.444820,6.813330",
			"video" : "rtsp://80.90.151.112/axis-media/media.amp"
		},
		"Western Cape" : 
		{
			"position" : "-33.925840,18.423220",
			"video" : "rtsp://196.21.92.82/axis-media/media.amp"
		},
		"Zeeland" : 
		{
			"position" : "51.5,3.613890",
			"video" : "rtsp://213.34.225.97/axis-media/media.amp"
		},
		"rtmp" : 
		{
			"video" : "rtmp://171.25.232.10/12d525bc9f014e209c1280bc0d46a87e"
		}
	}
}Logger level:4
HTTP Listen at 0.0.0.0:8000
 Requested URL : rtsp://97.68.104.34/axis-media/media.amp
Start playing sink for "video/H264" subsession
Close session: video/H264
`
[image]

mpromonet, 2023-08-10:
Hi @RezaFeng 

You might increase log level to see capture device discovery.

Best Regards,
Michel.

RezaFeng, 2023-08-10:
Sorry, here is the log
`|C--I|-|0|0|7205793488602807806|-]: Timed out after 15040 ms without a response, rtt=6000
[031:825][9] (p2p_transport_channel.cc:2113): Channel[0|1|__]: Removed connection 9801c4c0 (0 remaining)
[031:825][9] (p2p_transport_channel.cc:1902): Channel[0|1|__]: Transport channel state changed from 2 to 3
[031:825][9] (jsep_transport_controller.cc:1227): 0 Transport 1 state changed. Check if state is complete.
[031:825][9] (jsep_transport_controller.cc:1227): 0 Transport 1 state changed. Check if state is complete.
[031:825][8] (peer_connection.cc:1950): Changing IceConnectionState 1 => 4
[031:825][8] (PeerConnectionManager.h:287): virtual void PeerConnectionManager::PeerConnectionObserver::OnIceConnectionChange(webrtc::PeerConnectionInterface::IceConnectionState) state:failed peerid:0.03516571968366122
[031:826][8] (peer_connection.cc:1969): Changing standardized IceConnectionState 1 => 4
[031:826][66] (PeerConnectionManager.cpp:896): hangUp 0.03516571968366122
[031:826][66] (PeerConnectionManager.cpp:905): Remove PeerConnection peerid:0.03516571968366122
[031:826][66] (PeerConnectionManager.cpp:922): hangUp stream is no more used videocap0||
[031:826][66] (PeerConnectionManager.cpp:930): hangUp stream closed videocap0||
[031:826][6] (webrtc_video_engine.cc:1303): SetVideoSend (ssrc= 96034636, options: nullptr, source = nullptr)
[031:826][63] (resource_adaptation_processor.cc:129): Removing resource "QualityScalerResource".
[031:912][8] (PeerConnectionManager.h:278): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.03516571968366122
[031:912][66] (PeerConnectionManager.h:231): virtual PeerConnectionManager::PeerConnectionObserver::~PeerConnectionObserver()
[031:913][6] (webrtc_video_engine.cc:2191): VideoSendStream stats: 3802640, {input_fps: 13.0, encode_fps: 0, encode_ms: 0, encode_usage_perc: 0, target_bps: 0, media_bps: 0, suspended: false, bw_adapted_res: false, cpu_adapted_res: false, bw_adapted_fps: false, cpu_adapted_fps: false, #cpu_adaptations: 0, #quality_adaptations: 0} {ssrc: 96034636, type: media, width: 0, height: 0, key: 0, delta: 0, total_bps: 0, retransmit_bps: 0, avg_delay_ms: 0, max_delay_ms: 0, nack: 0, fir: 0, pli: 0}
[031:913][6] (webrtc_video_engine.cc:1443): Call stats: 3802640, {send_bw_bps: 0, recv_bw_bps: 0, max_pad_bps: 0, pacer_delay_ms: 0, rtt_ms: -1}
[031:914][8] (PeerConnectionManager.h:287): virtual void PeerConnectionManager::PeerConnectionObserver::OnIceConnectionChange(webrtc::PeerConnectionInterface::IceConnectionState) state:closed peerid:0.03516571968366122
[031:914][8] (sdp_offer_answer.cc:2808): Session: 5637063486585785061 Old state: stable New state: closed
[031:914][8] (PeerConnectionManager.h:284): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:closed peerid:0.03516571968366122
[031:914][6] (webrtc_video_engine.cc:2885): SetDefaultSink: nullptr
[031:914][6] (channel.cc:529): Channel disabled: {mid: 1, media_type: video}
[031:914][6] (channel.cc:1008): Changing video state, recv=0 send=0 for {mid: 1, media_type: video}
[031:914][63] (resource_adaptation_processor.cc:129): Removing resource "EncoderUsageResource".
[031:915][6] (video_send_stream_impl.cc:362): VideoSendStreamImpl::Stop
[031:916][6] (video_send_stream_impl.cc:309): ~VideoSendStreamImpl: {encoder_settings: { experiment_cpu_load_estimator: off}}, rtp: {ssrcs: [96034636], rids: [], mid: '1', rtcp_mode: RtcpMode::kReducedSize, max_packet_size: 1200, extmap-allow-mixed: true, extensions: [{uri: http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01, id: 3}, {uri: http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time, id: 2}, {uri: http://www.webrtc.org/experiments/rtp-hdrext/color-space, id: 8}, {uri: http://www.webrtc.org/experiments/rtp-hdrext/playout-delay, id: 5}, {uri: http://www.webrtc.org/experiments/rtp-hdrext/video-content-type, id: 6}, {uri: http://www.webrtc.org/experiments/rtp-hdrext/video-timing, id: 7}, {uri: urn:3gpp:video-orientation, id: 13}, {uri: urn:ietf:params:rtp-hdrext:sdes:mid, id: 4}, {uri: urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id, id: 11}, {uri: urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id, id: 10}], lntf: {enabled: false}, nack: {rtp_history_ms: 1000}, ulpfec: {ulpfec_payload_type: 118, red_payload_type: 116, red_rtx_payload_type: 117}, payload_name: VP8, payload_type: 96, raw_payload: false, flexfec: {payload_type: -1, ssrc: 0, protected_media_ssrcs: []}, rtx: {ssrcs: [2396258698], payload_type: 97}, c_name: BwwynQi0FLkoRfth}, rtcp_report_interval_ms: 1000, send_transport: (Transport), render_delay_ms: 0, target_delay_ms: 0, suspend_below_min_bitrate: off}
[031:916][6] (send_statistics_proxy.cc:672): WebRTC.Video.InputWidthInPixels 320
WebRTC.Video.InputHeightInPixels 240
WebRTC.Video.InputFramesPerSecond periodic_samples:7, {min:14, avg:15, max:15}
Frames encoded 0
WebRTC.Video.DroppedFrames.Capturer 0
WebRTC.Video.DroppedFrames.EncoderQueue 209
WebRTC.Video.DroppedFrames.Encoder 0
WebRTC.Video.DroppedFrames.Ratelimiter 0
WebRTC.Video.DroppedFrames.CongestionWindow 0
[031:916][6] (channel.cc:529): Channel disabled: {mid: 0, media_type: audio}
[031:916][6] (channel.cc:868): Changing voice state, recv=0 send=0 for {mid: 0, media_type: audio}
[031:916][9] (peer_connection.cc:2560): Tearing down data channel transport for mid=2
[031:917][8] (peer_connection.cc:568): Session: 5637063486585785061 is destroyed.
[031:918][66] (PeerConnectionManager.cpp:946): hangUp`

RezaFeng, 2023-08-10:
Oh, I find out what the problem is, I can you 127.0.0.1 to  get the stream, but I can not use my wifi IP, very strange. Could you tell me how to solve it?

mpromonet, 2023-08-10:
Hi,

maybe your firewall configuration ?

Best Regards,
Michel.

RezaFeng, 2023-08-10:
If I open the broswer on jetson nano, I can get the steam with 127.0.01 or 192.168.12.45(my wifi ip), the port is 8888, but if I use my own laptop, which connct to the same wifi, I can visit the website 192.168.12.45:8888, but I can not get the stream.
Here is the log when I use my laptop, I think it is the reason:
`[424:725][9] (network.cc:1053): Connect failed with 99
[424:741][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=6c457262752f4a6a3657436a, use_candidate=0, nomination=0
[424:789][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=63684237336c576349776a45, use_candidate=0, nomination=0
[424:838][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=574d6c623775623734615461, use_candidate=0, nomination=0
[424:886][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=656a306a4a2f2b7242505847, use_candidate=0, nomination=0
[424:934][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=455a5062674f586d63353355, use_candidate=0, nomination=0
[424:983][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=6d75706630457372456b3871, use_candidate=0, nomination=0
[425:031][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=426e797366342b2f42724a59, use_candidate=0, nomination=0
[425:079][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=30734b706a346d3963646a59, use_candidate=0, nomination=0
[425:128][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=6e2f6d707347396439625539, use_candidate=0, nomination=0
[425:176][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=666c494e347a6b457a74384a, use_candidate=0, nomination=0
[425:224][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=76494567446338556f636170, use_candidate=0, nomination=0
[425:273][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=586943456e38504762385179, use_candidate=0, nomination=0
[425:321][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=6f43794f516c505446673737, use_candidate=0, nomination=0
[425:370][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=61674a744f304b3061535267, use_candidate=0, nomination=0
[425:418][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=525531694558314c4e633338, use_candidate=0, nomination=0
[425:466][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=6d4e6d49654669424c4a7062, use_candidate=0, nomination=0
[425:515][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=65627a44534173754c2f4642, use_candidate=0, nomination=0
[... cut ...]
[425:756][9] (connection.cc:1441): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Sent STUN BINDING request, id=3369756f55593548767a6c75, use_candidate=0, nomination=0
[425:805][9] (connection.cc:909): Conn[ac020ed0:0:Net[eth0:172.17.0.x/16:Ethernet:id=1]:qOI86Smx:1:0:local:udp:172.17.0.x:56544->y1d7EJmc:1:1677729535:stun:udp:121.32.197.x:28233|C--I|-|0|0|7205793488602807806|-]: Timed out after 15042 ms without a response, rtt=6000
`

mpromonet, 2023-08-10:
Hi,

Ok you could reach the http server, then probably running docker with option --net=host will help you.

Best Regards,
Michel.

RezaFeng, 2023-08-10:
now I can visit the website, but it is an html page, how can I get the direct webrtc URL to embeding in my own html page?

mpromonet, 2023-08-10:
Hi,
There is différent way do go this, you Can use the webcomponent webrtc-streamer-element or use webrtc player using whep. There is notes in readme.
Best Regards
Michel

RezaFeng, 2023-08-11:
And how to set the width and height with docker if I use the V4L2 device？

closed 2024-05-18 by mpromonet