mpromonet/webrtc-streamer #619: No rule to make target '../webrtc/src/out/Release/obj/rtc_base/librtc_json.a', needed by 'webrtc-streamer'.

XingJinming-real, 2023-08-31:
When I build webrtc-streamer with make, it gives the following error.

[ 98%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtpvideocapturer.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspaudiocapturer.cpp.o
[ 99%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/rtspvideocapturer.cpp.o
make[2]: *** No rule to make target '../webrtc/src/out/Release/obj/rtc_base/librtc_json.a', needed by 'webrtc-streamer'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:124: CMakeFiles/webrtc-streamer.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

I tried [491](https://github.com/mpromonet/webrtc-streamer/issues/491), but it didn't work.

Thanks for your help

XingJinming-real, 2023-08-31:
I re-runed the command 
rm -rf ../webrtc/src/out && cmake . && make
and it gives me the error
[image]

mpromonet, 2023-08-31:
Did you don't install webrtc in ../webrtc ?

XingJinming-real, 2023-08-31:
[image]
[image]
this is the file structure.
I install webrtc through this command `fetch --no-history webrtc`

mpromonet, 2023-08-31:
In your log there is no build of webrtc, it should looks like :
```
...
CMAKE_CXX_COMPILER_ID=GNU
CMAKE_FIND_ROOT_PATH =
ALSA_FOUND = TRUE
PulseAudio_FOUND = 1
GTK3_FOUND = 1
Done. Made 1323 targets from 258 files in 1483ms
[1/3836] STAMP obj/poison_audio_codecs.stamp
[2/3836] STAMP obj/poison_default_task_queue.stamp
...
``` 
maybe you have not permission to remove ../webrtc/src/out ? in order to rebuild it....

XingJinming-real, 2023-08-31:
[image]
I removed the out dir and it gives me this

mpromonet, 2023-08-31:
Then you need to debug why gn gen & ninja are not run between GTK_FOUND & RMTP_FOUND

XingJinming-real, 2023-08-31:
I figured it out.
I didn't install ninja and pkg-config
Thanks a lot

AndySkaura, 2023-10-23:
The same problem occurred, and I don't know what happened, but during the _cmake ._ stage, the following was encountered:
`CMAKE_BUILD_TYPE = Release
WEBRTCROOT = /home/webrtc-streamer/../webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=GNU
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMAKE_FIND_ROOT_PATH = 
ALSA_FOUND = FALSE
PulseAudio_FOUND = 0
GTK3_FOUND = 
ERROR at //build/config/linux/pkg_config.gni:104:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "json")
                ^----------
Current dir: /home/webrtc/src/out/Release/
Command: python3 /home/webrtc/src/build/config/linux/pkg-config.py glib-2.0 gmodule-2.0 gobject-2.0 gthread-2.0
Returned 1.
stderr:

Package glib-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `glib-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'glib-2.0' found
Package gmodule-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gmodule-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gmodule-2.0' found
Package gobject-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gobject-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gobject-2.0' found
Package gthread-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gthread-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gthread-2.0' found
Could not run pkg-config.
`
Therefore, I installed _sudo apt-get install libglib2.0-dev_, and it started working correctly.
I hope I can be helpful to you.

closed 2023-12-30 by mpromonet