mpromonet/webrtc-streamer #620: Have you found a browser that supports H265 playback?

liusuyi2021, 2023-09-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [windows]
 - Browser [chrome,]
 - Version [0.8.3]

**Additional context**
Have you found a browser that supports H265 playback? Google and Edge already support HEVC decoding, but I am unable to decode H265 from HIKVISION cameras after webrtc streamer transcoding. May I ask if it is a camera issue or a webrtc streamer issue

mpromonet, 2023-09-06:
Hi,

No I didn't, it seems only Safari 14+ support it https://github.com/AlexxIT/go2rtc#codecs-madness ... 

Best Regards,
Michel.

liusuyi2021, 2023-09-15:
> 晉語
> 
> 不，我没有，似乎只有 Safari 14+ 支持它 https://github.com/AlexxIT/go2rtc#codecs-madness......
> 
> 最好的问候，米歇尔。

I'm calling Kara again. Is it possible to solve the problem of playing H265 on Google Chrome

mpromonet, 2023-12-30:
you may look to https://bugs.chromium.org/p/webrtc/issues/detail?id=15725&q=h.265&can=2

mpromonet, 2026-10-18:
Hi,
I am closing this as it is working, latest with -o option on chrome 155 is displaying h265,
Best Regards,
Michel.

closed 2026-10-18 by mpromonet