mpromonet/webrtc-streamer #622: encountering fatal errors when cmake .

XingJinming-real, 2023-09-10:
(base) wisn@gpu:~/webrtc-streamer$ cmake .
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMAKE_BUILD_TYPE = Release
WEBRTCROOT = /home/wisn/webrtc-streamer/../webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=GNU
-- Found Git: /usr/bin/git (found version "2.34.1")
CMAKE_FIND_ROOT_PATH =
ALSA_FOUND = FALSE
PulseAudio_FOUND = 0
GTK3_FOUND = 1
Done. Made 1323 targets from 258 files in 857ms
[1/3825] CXX obj/api/media_stream_interface/media_stream_interface.o
FAILED: obj/api/media_stream_interface/media_stream_interface.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/api/media_stream_interface/media_stream_interface.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-17-init-16420-g0c545a44-8\" -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_STRICT_FIELD_TRIALS=0 -DRTC_ENABLE_VP9 -DRTC_DAV1D_IN_INTERNAL_DECODER_FACTORY -DWEBRTC_HAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_LIBRARY_IMPL -DWEBRTC_ENABLE_AVX2 -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -DLIBYUV_DISABLE_NEON -I../.. -Igen -I../../third_party/abseil-cpp -I../../third_party/libyuv/include -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -m64 -msse3 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -ffile-compilation-dir=. -no-canonical-prefixes -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -Wglobal-constructors -Wno-shadow -Wctad-maybe-unsupported -Wc++11-narrowing -Wundef -Wunused-lambda-capture -std=c++17 -Wno-trigraphs -gsimple-template-names -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../../api/media_stream_interface.cc -o obj/api/media_stream_interface/media_stream_interface.o
In file included from ../../api/media_stream_interface.cc:11:
../../api/media_stream_interface.h:21:10: fatal error: 'string' file not found
   21 | #include <string>
      |          ^~~~~~~~
1 error generated.
[2/3825] CXX obj/api/rtc_error/rtc_error.o
FAILED: obj/api/rtc_error/rtc_error.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/api/rtc_error/rtc_error.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-17-init-16420-g0c545a44-8\" -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_STRICT_FIELD_TRIALS=0 -DRTC_ENABLE_VP9 -DRTC_DAV1D_IN_INTERNAL_DECODER_FACTORY -DWEBRTC_HAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_LIBRARY_IMPL -DWEBRTC_ENABLE_AVX2 -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I../.. -Igen -I../../third_party/abseil-cpp -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -m64 -msse3 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -ffile-compilation-dir=. -no-canonical-prefixes -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -Wglobal-constructors -Wno-shadow -Wctad-maybe-unsupported -Wc++11-narrowing -Wundef -Wunused-lambda-capture -std=c++17 -Wno-trigraphs -gsimple-template-names -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../../api/rtc_error.cc -o obj/api/rtc_error/rtc_error.o
In file included from ../../api/rtc_error.cc:11:
../../api/rtc_error.h:17:10: fatal error: 'string' file not found
   17 | #include <string>
      |          ^~~~~~~~
1 error generated.
[3/3825] CXX obj/api/rtc_event_log_output_file/rtc_event_log_output_file.o
FAILED: obj/api/rtc_event_log_output_file/rtc_event_log_output_file.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/api/rtc_event_log_output_file/rtc_event_log_output_file.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-17-init-16420-g0c545a44-8\" -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_STRICT_FIELD_TRIALS=0 -DRTC_ENABLE_VP9 -DRTC_DAV1D_IN_INTERNAL_DECODER_FACTORY -DWEBRTC_HAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_LIBRARY_IMPL -DWEBRTC_ENABLE_AVX2 -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I../.. -Igen -I../../third_party/abseil-cpp -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -m64 -msse3 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -ffile-compilation-dir=. -no-canonical-prefixes -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -Wglobal-constructors -Wno-shadow -Wctad-maybe-unsupported -Wc++11-narrowing -Wundef -Wunused-lambda-capture -std=c++17 -Wno-trigraphs -gsimple-template-names -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../../api/rtc_event_log_output_file.cc -o obj/api/rtc_event_log_output_file/rtc_event_log_output_file.o
In file included from ../../api/rtc_event_log_output_file.cc:11:
../../api/rtc_event_log_output_file.h:17:10: fatal error: 'string' file not found
   17 | #include <string>
      |          ^~~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
[... cut ...]
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
OpenSSL_FOUND = TRUE
-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC
-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC - Success
-- Performing Test CXX_SUPPORTS_STATIC_LIBSTDCPP
-- Performing Test CXX_SUPPORTS_STATIC_LIBSTDCPP - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wisn/webrtc-streamer

When I run cmake . it gives above error. 
After search, I installed libc++-dev, but it sill gives this error. How should I do to fix it
Thanks.

mpromonet, 2023-09-10:
Hi,

You may look to Dockerfile to see what is needed in the build environment.

Best Regards,
Michel.

XingJinming-real, 2023-09-10:
[image]

I installed the packages mentioned in the dockerfile, but it still gives the same error.  
gcc and g++ are installed.

mpromonet, 2023-09-11:
You installed pkgs for runtime, pkgs for build are in build stage.

closed 2023-12-30 by mpromonet