mpromonet/webrtc-streamer #623: need help 用代码模拟了个rtsp源，如何告诉你的程序这个rtsp地址不存在(Simulated an rtsp source with code, how to tell your program that the rtsp address does not exist)

yangjieshao, 2023-09-21:
由于原始的rtsp地址带有账号密码
(Because the original rtsp address contains the account password)
所以我用c#写了个程序来接收和转发rtsp数据
(So I wrote a program in C # to receive and forward rtsp data)

当我发现这个地址不存在的时候，我会直接关闭流
(When I discover that this address does not exist, I will directly close the stream)
```

while (!CancellationTokenSource.IsCancellationRequested)
{
	try
	{
		TcpClient src = Server.AcceptTcpClient();
		NetworkStream ns1 = src.GetStream();
		byte[] bt = new byte[10240];
		int count = ns1.Read(bt, 0, bt.Length);
		string str = Encoding.ASCII.GetString(bt, 0, count);
		var srcTransportInfo = GetTransportInfo(str);
		if (srcTransportInfo == null)
		{
// here !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
			Logger.LogInformation("未找到目标IP地址，关闭连接");
			src.Close();
			src.Dispose();
			continue;
		}
// TODO 
// Normal data processing
// …………………………
		TcpClient dst = new(srcTransportInfo.DstIP, srcTransportInfo.DstPort);

		//设定超时，否则端口将一直被占用，即使失去连接
		src.SendTimeout = 300000;
		src.ReceiveTimeout = 300000;
		dst.SendTimeout = 300000;
		dst.ReceiveTimeout = 300000;
		srcTransportInfo.Src = src;
		srcTransportInfo.Dst = dst;
		var dstTransportInfo = new TransportInfo()
		{
			Src = srcTransportInfo.Dst,
			Dst = srcTransportInfo.Src,
			NeedReplace = false
		};

		Task.Factory.StartNew(() => { Transfer(srcTransportInfo); });
		Task.Factory.StartNew(() => { Transfer(dstTransportInfo); });

		byte[] buf = Encoding.ASCII.GetBytes(str.Replace(srcTransportInfo.SrcAddress, srcTransportInfo.DstAddress));
		dst.GetStream().Write(buf, 0, buf.Length);
		Logger.LogInformation($"TCP发起连接 {srcTransportInfo.SrcAddress} ==>> {srcTransportInfo.DstAddress} 已建立连接");
	}
	catch { continue; }
}
```

但是现在发现，直接断开流，你的程序会不断尝试重连
(But now I find that if you disconnect the stream directly, your program will keep trying to reconnect.)

如果是传了个不经过我的程序的不存在的rtsp地址
(If it was a non-existent rtsp address that did not pass through my program)
你的程序会正常的断开链接
(Your program will disconnect normally)

所以 我该如何告诉你的程序，这个rtsp地址不存在，不要再发起重连请求了?
(So how should I tell your program that this rtsp address does not exist and do not initiate reconnection requests?)

mpromonet, 2023-09-21:
Hi,
If you do not open a webrtc connection for the rtsp URL you don't want to connect, it will not try to connect. But i am not to understand what you are trying to achieve.
Webrtc-streamer don't need to know backend URL, when a webrtc call is initiated, backend URL is given as argument.
Best Regards
Michel

yangjieshao, 2023-09-22:
for now
[image]
everyone can see **username** and **password** in url
some customers do not allow this situation

**and I don't know how to c++**
so  I add a program by c#
[image]

But I cannot guarantee that the virtual rtsp address passed has a real corresponding rtsp address
So how do I tell webrtc-streamer when I receive a virtual address that has no real corresponding address?

mpromonet, 2023-09-22:
Hi,
If you put a name and an rtsp URL with user&password in the config file or using arguments -n name and -u URL, only name is known by the web browser. Then user, password, IP, ... are private to the backend.
Best Regards
Michel

yangjieshao, 2023-09-23:
使用配置文件的话，该如何实时响应用户添加摄像头/修改摄像头账号密码呢？
(If a configuration file is used, how to respond in real time to the user adding a camera/modifying the camera account password?)

closed 2024-05-18 by mpromonet