mpromonet/webrtc-streamer #625: How to stream desktop and local mkv video files?

XingJinming-real, 2023-09-25:
I don't know how to config the conf.json to make webrtc-streamer stream desktop captures and local video files. Below is my config file. I want to know how to modify it to make it work.
[image]
Thanks very much for your help.

XingJinming-real, 2023-09-25:
Machine:
CPU:intel i7-11800h
GPU:rtx-3060 laptop

Regarding local video streaming, I change the dir of the video, putting it to the root dir of the webrtc-streamer. Then the local video streaming works, but in a very leggy way, losing many frames. 

And the screen capture still doesn't work.

Could you tell me how to improve the smoothness of the local video streaming and how to do screen capture.

Thanks!

mpromonet, 2023-10-08:
Hi,

screen:// and window:// scheme are using names of the screen or window title to stream. You should be able to get this from /api/getMediaList if it was build with screen capture. 
It needs to have WEBRTCDESKTOPCAPTURE set to ON (it is default) and need to have gtk3 installed.

Best Regards,
Michel.

closed 2025-05-17 by mpromonet