mpromonet/webrtc-streamer #628: It is invalid。callurl += "&options=" + encodeURIComponent({"width":"320","height":"240"});

swj369451, 2023-10-19:
It is invalid。callurl += "&options=" + encodeURIComponent({"width":"320","height":"240"});
I received the stream is frameWidth | 704 frameHeight | 576

mpromonet, 2023-10-19:
Hi @swj369451 

What is invalid ? It may depends on the source.
If you describe your problem you may get better informations.

Best Regards,
Michel.

swj369451, 2023-10-21:
I want to modify the quality of the output video to reduce the bandwidth.
I see the option parameter.But I don't know what specific parameters can be configured to reduce the image quality.
thank you for answering my question.

mpromonet, 2023-10-21:
Hi,

I guess you should use something like `"&options=" + encodeURIComponent("width=320&height=240")`

Best Regards,
Michel.

swj369451, 2023-10-22:
Hi,
I configured option, but it didn't work, the resolution didn't change

callurl is http://xxx.xxx.xxx.xxx:7502/api/call?peerid=0.03189659933396083&url=rtsp%3A%2F%2Fadmin%3Aa1234567%40192.168.2.3%3A554%2Fh264%2Fch33%2Fsub%2Fav_stream&options=width%3D320%26height%3D240.

thank you for answering my question.

mpromonet, 2023-10-22:
Hi,

If you are using '-o' to forward encoded packet, the resolution cannot be changed. 

Best Regards,
Michel.

swj369451, 2023-10-22:
I didn't use "-o"

webrtc-streamer.exe -H 127.0.0.1:9001 -C config.json -Sxxx.xxx.xxx.xxx:3478 -Ttest:123@xxx.xxx.xxx.xxx:3478

thank you for answering my question.

mpromonet, 2023-10-22:
Hi,

I guess you have already open the backend stream, and then the resolution is decided by the first stream. You could check with admin.html page.

Best Regards,
Michel.

closed 2024-01-20 by mpromonet