mpromonet/webrtc-streamer #630: Library not loaded: /usr/local/opt/rtmpdump/lib/librtmp.1.dylib

sashangli, 2023-10-24:
Library not loaded: /usr/local/opt/rtmpdump/lib/librtmp.1.dylib
  Referenced from: <14618B62-D8DD-3191-BE74-B702F824B31C> /Users/lizhuohao/zhangqi/app/webrtc-streamer-v0.8.3-Darwin-x86_64-Release/webrtc-streamer
  Reason: tried: '/usr/local/opt/rtmpdump/lib/librtmp.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/rtmpdump/lib/librtmp.1.dylib' (no such file), '/usr/local/opt/rtmpdump/lib/librtmp.1.dylib' (no such file), '/usr/local/lib/librtmp.1.dylib' (no such file), '/usr/lib/librtmp.1.dylib' (no such file, not in dyld cache)
zsh: abort

mpromonet, 2023-10-24:
Hi,
I Guess you should install this library, or build without rtmp support.
Best Regards
Michel

sashangli, 2023-10-25:
thanks，run `brew install rtmpdump`，then success

closed 2023-10-28 by mpromonet