mpromonet/webrtc-streamer #632: docker container return [h264 @ 0x7f682b0ed000] no frame!

LRxRayTse, 2023-11-01:
Hi, I want to ask why container console keep showing following message. (The steaming is still working)
 
[image]

Thank you. <3

yangjieshao, 2023-11-01:
try 
```
docker run --restart=always -d --network=host \
           --name webrtc-streamer \
           mpromonet/webrtc-streamer 
```

only 8000 is not enough

mpromonet, 2023-11-01:
Hi @LRxRayTse 

This is coming from h264 decoder of ffmeg that cannot get frame, this depends on your backend stream. Probably ffplay with the same backend stream will do the same....

Best Regards,
Michel.

LRxRayTse, 2023-11-02:
@yangjieshao @mpromonet 

Thank a lot, I will try and check my backend stream. <3

closed 2024-05-18 by mpromonet