mpromonet/webrtc-streamer #634: Http Authorization Failed

mjchow, 2023-11-03:
Hello, thank you for providing such a good project. I have a question now. I configured the content in password.txt, for example: test:127.0.0.1:123456. But when I browse 127.0.0.1:8000, it requires me to provide a username and password. When I enter test and 123456, the authentication fails. Can you please tell me what should I do?

mpromonet, 2023-11-03:
Hi,

The description of the file format is https://github.com/civetweb/civetweb/blob/master/docs/UserManual.md#global_auth_file.
You should not put the password in plain text but the md5 using your realm.

Best Regards,
Michel.

mjchow, 2023-11-06:
> Hi,
> 
> The description of the file format is https://github.com/civetweb/civetweb/blob/master/docs/UserManual.md#global_auth_file. You should not put the password in plain text but the md5 using your realm.
> 
> Best Regards, Michel.

May I ask what is the generation rule for this password? MD5(password) or MD5(realm+password)? I have tried several options, but none of them were successful in HTTP authentication.

mpromonet, 2023-11-18:
Hi,
The realm is the authentification domain which is by default mydomain.com
```
echo user:mydomain.com:$(echo -n user:mydomain.com:password | md5sum | cut -d- -f1) > passwd 
webrtc-streamer -A passwd
```
Best Regards,
Michel.

closed 2023-12-30 by mpromonet