mpromonet/webrtc-streamer #637: No start code is found

MHLiu1219, 2023-11-07:
Hello, I'm having some problems using the web to play RTSP video streams. I was able to play it normally with VLC Player, but it seems that the video cannot be decoded properly when using the web. And there are some save logs:

[image]
[image]
[image]

mjchow, 2023-11-07:
I have also encountered similar problems.

mpromonet, 2023-11-07:
Hi,

It seems that H264 frame are not correctly proceeded in your case, it is probably in https://github.com/mpromonet/webrtc-streamer/blob/master/inc/livevideosource.h. Increasing log level may helps you to understand the problem.

Best Regards,
Michel.

buithikieu, 2024-03-29:
> Hi,
> 
> It seems that H264 frame are not correctly proceeded in your case, it is probably in https://github.com/mpromonet/webrtc-streamer/blob/master/inc/livevideosource.h. Increasing log level may helps you to understand the problem.
> 
> Best Regards, Michel.

How to increase log level, please help me