mpromonet/webrtc-streamer #638: web player could not start but keep waiting

shaojun, 2023-11-09:
**Describe the bug**
this is my server side running info, every time I freshed the client machine's browser page, the line of `Requested URL :......Start playing sink ...` will be printed:
```
aaaa@aaaa:~/Download/webrtc-streamer-v0.8.4-Linux-armv7l-Release$ ./webrtc-streamer -C config.json
Version:v0.8.4/Linux-armv7l civetweb@v1.16 webrtc@507f1cc-dirty live555helper@0e74999
{
        "urls" :
        {
                "Ele" :
                {
                        "video" : "rtsp://aaa:aaa@192.168.177.3:554/h264/ch2/main/av_stream"
                }
        }
}Logger level:4
HTTP Listen at 0.0.0.0:8000
 Requested URL : rtsp://aaa:aaa@192.168.177.3:554/h264/ch2/main/av_stream
Start playing sink for "video/H264" subsession

```

and this is my web browser that opened web page: `/html/webrtcstreamer.html?video=Ele`, but it's keep rolling:
[image]

and this is the browser's debug console info:

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Debian, linux arm v7. 
 - Browser [e.g. chrome, safari] chrome, Edge
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mpromonet, 2023-11-14:
Hi,

It seems ICE negociation fails, you may try to use another STUN server that the default stun.l.google.com:19302. You may try to use the internal one adding `-S ` argument.

Best Regards,
Michel.

stlobe, 2023-11-21:
Hi Michel, I'm having the same problem. However, this problem occurs with some Internet service providers. I tried setting up a STUN server and setting the server to DMZ, but I still have this issue. Do you have any ideas?

mpromonet, 2023-11-22:
Hi @stlobe 

You may try to use a TURN server https://gabrieltanner.org/blog/turn-server/ or others.

Best Regards,
Michel.

stlobe, 2023-11-23:
Thanks Michel, I figured out it was a udp port issue.  I tried adding udp scopes on both internal and external firewalls and it works now.