mpromonet/webrtc-streamer #639: rtsp://admin:a4451442@192.168.23.202/h265/ch22/main/av_stream 在html中如何获取音频

user0568, 2023-11-14:
webRtcServer4 = new WebRtcStreamer("video4","http://127.0.0.1:8000");
webRtcServer4.connect("rtsp://admin:a4451442@192.168.23.202/h265/ch22/main/av_stream");

如何获取音频

mpromonet, 2023-11-14:
Bonjour @user0568,

Je vous remercie pour votre message, mais ne comprenant pas ce que vous dites, il ne m'est pas possible de vous répondre.

Bien cordialement
Michel

user0568, 2023-11-14:
<!DOCTYPE html>
<html>
	<head>
		<meta charset="utf-8">
		<title></title>
<script src="html/libs/adapter.min.js" ></script>
		<script src="html/webrtcstreamer.js" ></script>
		<script>        
		    var webRtcServer = null;
		    //页面加载时加载视频画面
		    window.onload = function() { 
		    	//video：需要绑定的video控件ID
		    	//192.168.1.226：启动webrtc-streamer的设备IP
		        webRtcServer = new WebRtcStreamer("video","http://192.168.100.17:8000");
		        //需要查看的rtsp地址
				webRtcServer.connect("rtsp://admin:1242@192.168.23.17/mpeg4/ch33/sub/av_stream");
		    }  
			//页面退出时销毁
		    window.onbeforeunload = function() { 
				webRtcServer.disconnect();				
			}
		</script>
	</head>
	<body>
		<video id='video' style='object-fit:fill' controls autoplay autobuffer muted preload='auto' width="600px" height="300px" audio="true"></video>
	</body>
</html>

The front-end page uses the above code, but only the video is played without audio. I use vlc to play both the video and audio. May I ask if WebRtcStreamer cannot decode the audio encoding in the stream

Learners

user0568, 2023-11-14:
> 卓悦 ，
> 
> Je vous remercie pour votre message， mais ne comprenant pas ce que vous dites， il ne m'est pas possible de vous répondre.
> 
> Bien cordialement 米歇尔
[image]

The front-end page uses the above code, but only the video is played without audio. I use vlc to play both the video and audio. May I ask if WebRtcStreamer cannot decode the audio encoding in the stream

Learners

mpromonet, 2023-11-14:
Hi @user0568,

With this code, you will get only video, if you like to get video & audio, you need to give an audio url.
You might try 
```
webRtcServer.connect("rtsp://admin:a4451442@192.168.23.202/h265/ch22/main/av_stream","rtsp://admin:a4451442@192.168.23.202/h265/ch22/main/av_stream");
```

Best Regards,
Michel.

closed 2024-01-20 by mpromonet

richard202205, 2024-03-25:
> Hi @user0568,
>
> With this code, you will get only video, if you like to get video & audio, you need to give an audio url. You might try
>
> ``` [...]

I did what you said but there's still no audio playing. :(

Gakki-yn, 2024-12-13:
use 0.8.3