mpromonet/webrtc-streamer #640: How to use Let's Encrypt SSL to start webrtc-streamer?

stlobe, 2023-11-20:
Hi All,

I installed Let's Encrypt SSL certificate and tried to start the service but getting Segment Failed. All certificates are stored in 'ssl' directory.
The command:
webrtc-streamer -C config.json -H443s -c ssl/

It seems the problem is occurred by -H443s. I tried to remove 's' and it starts without SSL. 
Can anyone help?

Thank you

mpromonet, 2023-11-20:
Hi,

This was discussed https://github.com/mpromonet/webrtc-streamer/discussions/618, you should pass to -c  a file that concatenate certificate and key (not a folder).

Best Regards,
Michel.

stlobe, 2023-11-21:
Thank you Michel. But it still prompted ' Segmentation fault'.

stlobe, 2023-11-21:
Oh. I found the problem that was caused by openssl devep lib. The problem is solved. Thank Michel again.

closed 2023-11-21 by stlobe