mpromonet/webrtc-streamer #641: If the public ip address is deployed, the connection cannot be connected and the video cannot be turned on

long255, 2023-11-22:
Get IceServers
webrtcstreamer.js:163 createPeerConnection  config: {"iceServers":[{"urls":["stun:stun.l.google.com:19302"]}],"iceTransportPolicy":"all"}
webrtcstreamer.js:222 Created RTCPeerConnnection with config: {"iceServers":[{"urls":["stun:stun.l.google.com:19302"]}],"iceTransportPolicy":"all"}
webrtcstreamer.js:203 codecs:[]
webrtcstreamer.js:239 End of candidates.
webrtcstreamer.js:256 Remote track added:{"isTrusted":true}
webrtcstreamer.js:276 setRemoteDescription ok
6webrtcstreamer.js:248 addIceCandidate ok:true
webrtcstreamer.js:298 Adding ICE candidate :{"candidate":"candidate:1138562957 1 udp 2122260223 172.20.0.1 24054 typ host generation 0 ufrag 9E3u network-id 1 network-cost 50","sdpMid":"0","sdpMLineIndex":0,"usernameFragment":""}
webrtcstreamer.js:298 Adding ICE candidate :{"candidate":"candidate:2206320160 1 udp 2122194687 172.18.0.1 62007 typ host generation 0 ufrag 9E3u network-id 2 network-cost 50","sdpMid":"0","sdpMLineIndex":0,"usernameFragment":""}
webrtcstreamer.js:298 Adding ICE candidate :{"candidate":"candidate:2480297516 1 udp 2122129151 172.17.0.1 38054 typ host generation 0 ufrag 9E3u network-id 3 network-cost 50","sdpMid":"0","sdpMLineIndex":0,"usernameFragment":""}
webrtcstreamer.js:298 Adding ICE candidate :{"candidate":"candidate:4080837569 1 udp 2122063615 10.0.0.15 54752 typ host generation 0 ufrag 9E3u network-id 4","sdpMid":"0","sdpMLineIndex":0,"usernameFragment":""}
webrtcstreamer.js:298 Adding ICE candidate :{"candidate":"candidate:1024652565 1 tcp 1518280447 172.20.0.1 45182 typ host tcptype passive generation 0 ufrag 9E3u network-id 1 network-cost 50","sdpMid":"0","sdpMLineIndex":0,"usernameFragment":""}
webrtcstreamer.js:298 Adding ICE candidate :{"candidate":"candidate:4249756856 1 tcp 1518214911 172.18.0.1 50849 typ host tcptype passive generation 0 ufrag 9E3u network-id 2 network-cost 50","sdpMid":"0","sdpMLineIndex":0,"usernameFragment":""}
webrtcstreamer.js:298 Adding ICE candidate :{"candidate":"candidate:3977884852 1 tcp 1518149375 172.17.0.1 37883 typ host tcptype passive generation 0 ufrag 9E3u network-id 3 network-cost 50","sdpMid":"0","sdpMLineIndex":0,"usernameFragment":""}
webrtcstreamer.js:298 Adding ICE candidate :{"candidate":"candidate:2381529433 1 tcp 1518083839 10.0.0.15 48806 typ host tcptype passive generation 0 ufrag 9E3u network-id 4","sdpMid":"0","sdpMLineIndex":0,"usernameFragment":""}
webrtcstreamer.js:171 oniceconnectionstatechange  state: checking
8webrtcstreamer.js:299 addIceCandidate OK
webrtcstreamer.js:171 oniceconnectionstatechange  state: disconnected

long255, 2023-11-22:
Is in the LAN surveillance video can be played out, but deployed to the public ip server, the video can not be played out

w419894293, 2023-11-24:
在外网使用时需要使用Turn服务来进行打洞

long255, 2023-11-24:
非常感谢，现在可以了

w419894293, 2023-11-24:
您的邮件已经收到，稍后会处理.