mpromonet/webrtc-streamer #642: Can I use the Apache HTTP server to replace the internal HTTP server?

stlobe, 2023-11-22:
> I tried to install the Apache server and run the sample code without any reason. Do I need to start the webrtc-streamer binary file and activate port 8000 first?

Thanks

mpromonet, 2023-11-22:
Hi @stlobe 

I am not sur to understand, you could use whatever HTTP server you like, an example is describe in https://github.com/mpromonet/webrtc-streamer#html-embedding

Best Regards,
Michel.

stlobe, 2023-11-23:
Oh sorry Michel...
I tried installing the Apache server and running the embed sample code, but nothing happened.  Do I need to start the webrtc-streamer program first to activate port 8000?

stlobe, 2023-11-23:
The cose is finally worked on Apache. Thank you Michel

closed 2023-11-23 by stlobe