mpromonet/webrtc-streamer #646: Failed to DESCRIBE: URL does not begin with

zhaozhenshun, 2023-11-24:
docker run laster version.  report this question. but this address is ok

Failed to DESCRIBE: URL does not begin with "rtsp://" or "rtsps://"
[1019:776][167] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://97.68.104.34/axis-media/media.amp error:URL does not begin with "rtsp://" or "rtsps://"

system 
[image]
[image]

mpromonet, 2023-11-27:
Hi,

I just try with latest version it seems to works well:
`mpromonet/webrtc-streamer                                                                  latest                    d4b12ad570d6   8 days ago      267MB
`
Best Regards,
Michel.

mpromonet, 2023-11-30:
Hi,

What webrtc stream are you trying to call ? Probably increasing log level will help you.

Best Regards
Michel

hk1993, 2023-11-30:
1.use  this to run
docker run -e "LOG_LEVEL=debug" -itd -p 28000:8000 --name webrtc-streamer  mpromonet/webrtc-streamer
bug log is same up
2.click demo buttons  for  webrtc stream trying, then log show "Failed to DESCRIBE: URL does not begin with "rtsp://" or "rtsps://""
demo buttons like  "Allendale"、"Asahi"....

closed 2025-01-19 by mpromonet