mpromonet/webrtc-streamer #650: for rtsp stream generate by ffmpeg ,can't read the rtsp session

KevinYang123123, 2023-12-11:
I have built several rtsp streams by ffmpeg like this command : ffmpeg -re -stream_loop -1  -i /home/kevin/app/a.mp4 -an -c:v libx264  -f rtsp  rtsp://localhost:8555/people/fall_detect
here is the log:
```
Received 87 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 461 Unsupported Transport
CSeq: 3
Server: gortsplib
Session: 1825256676

Failed to SETUP: 461 Unsupported Transport
[147:722][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:461 Unsupported Transport
Failed to PLAY: No RTSP session is currently in progress

[147:722][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:No RTSP session is currently in progress

[148:607][10] (network.cc:1053): Connect failed with 99
Failed to TEARDOWN: No RTSP session is currently in progress

[148:722][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:No RTSP session is currently in progress

Created new TCP socket 10 for connection
Connecting to 192.168.50.186, port 8555 on socket 10...
...remote connection opened
Sending request: DESCRIBE rtsp://192.168.50.186:8555/people/fall_detect RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2023.11.30
Accept: application/sdp

Received 419 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
CSeq: 2
Content-Base: rtsp://192.168.50.186:8555/people/fall_detect/
Content-Length: 258
Content-Type: application/sdp
Server: gortsplib

v=0
o=- 0 0 IN IP4 127.0.0.1
s=Stream
c=IN IP4 0.0.0.0
t=0 0
m=video 0 RTP/AVP 96
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1; sprop-parameter-sets=Z2QAKKzZQHgCJ+WEAAADAAQAAAMAyDxgxlg=,aOvjyyLA; profile-level-id=640028
a=control:trackID=0

Sending request: SETUP rtsp://192.168.50.186:8555/people/fall_detect/trackID=0 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2023.11.30
Transport: RTP/AVP;unicast;client_port=56334-56335

Received 87 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 461 Unsupported Transport
CSeq: 3
Server: gortsplib
Session: 2666636558

Failed to SETUP: 461 Unsupported Transport
[148:724][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:461 Unsupported Transport
Failed to PLAY: No RTSP session is currently in progress

[148:724][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:No RTSP session is currently in progress

Failed to TEARDOWN: No RTSP session is currently in progress

[149:724][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:No RTSP session is currently in progress

Created new TCP socket 10 for connection
Connecting to 192.168.50.186, port 8555 on socket 10...
...remote connection opened
Sending request: DESCRIBE rtsp://192.168.50.186:8555/people/fall_detect RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2023.11.30
Accept: application/sdp

Received 419 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
CSeq: 2
Content-Base: rtsp://192.168.50.186:8555/people/fall_detect/
Content-Length: 258
Content-Type: application/sdp
Server: gortsplib

v=0
o=- 0 0 IN IP4 127.0.0.1
s=Stream
c=IN IP4 0.0.0.0
t=0 0
m=video 0 RTP/AVP 96
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1; sprop-parameter-sets=Z2QAKKzZQHgCJ+WEAAADAAQAAAMAyDxgxlg=,aOvjyyLA; profile-level-id=640028
a=control:trackID=0

Sending request: SETUP rtsp://192.168.50.186:8555/people/fall_detect/trackID=0 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2023.11.30
Transport: RTP/AVP;unicast;client_port=35534-35535

Received 87 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 461 Unsupported Transport
CSeq: 3
Server: gortsplib
Session: 2668601755

Failed to SETUP: 461 Unsupported Transport
[149:725][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:461 Unsupported Transport
Failed to PLAY: No RTSP session is currently in progress

[149:725][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:No RTSP session is currently in progress

[150:607][10] (network.cc:1053): Connect failed with 99
Failed to TEARDOWN: No RTSP session is currently in progress

[150:726][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:No RTSP session is currently in progress

Created new TCP socket 10 for connection
Connecting to 192.168.50.186, port 8555 on socket 10...
...remote connection opened
Sending request: DESCRIBE rtsp://192.168.50.186:8555/people/fall_detect RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2023.11.30
Accept: application/sdp

Received 419 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
CSeq: 2
Content-Base: rtsp://192.168.50.186:8555/people/fall_detect/
Content-Length: 258
Content-Type: application/sdp
Server: gortsplib

v=0
o=- 0 0 IN IP4 127.0.0.1
s=Stream
c=IN IP4 0.0.0.0
t=0 0
m=video 0 RTP/AVP 96
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1; sprop-parameter-sets=Z2QAKKzZQHgCJ+WEAAADAAQAAAMAyDxgxlg=,aOvjyyLA; profile-level-id=640028
a=control:trackID=0

Sending request: SETUP rtsp://192.168.50.186:8555/people/fall_detect/trackID=0 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2023.11.30
Transport: RTP/AVP;unicast;client_port=52666-52667

Received 86 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 461 Unsupported Transport
CSeq: 3
Server: gortsplib
Session: 947755586

Failed to SETUP: 461 Unsupported Transport
[150:726][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:461 Unsupported Transport
Failed to PLAY: No RTSP session is currently in progress

[150:726][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:No RTSP session is currently in progress

Failed to TEARDOWN: No RTSP session is currently in progress

[151:726][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:No RTSP session is currently in progress

Created new TCP socket 10 for connection
Connecting to 192.168.50.186, port 8555 on socket 10...
...remote connection opened
Sending request: DESCRIBE rtsp://192.168.50.186:8555/people/fall_detect RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2023.11.30
Accept: application/sdp

Received 419 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
CSeq: 2
Content-Base: rtsp://192.168.50.186:8555/people/fall_detect/
Content-Length: 258
Content-Type: application/sdp
Server: gortsplib

v=0
o=- 0 0 IN IP4 127.0.0.1
s=Stream
c=IN IP4 0.0.0.0
t=0 0
m=video 0 RTP/AVP 96
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1; sprop-parameter-sets=Z2QAKKzZQHgCJ+WEAAADAAQAAAMAyDxgxlg=,aOvjyyLA; profile-level-id=640028
a=control:trackID=0

Sending request: SETUP rtsp://192.168.50.186:8555/people/fall_detect/trackID=0 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2023.11.30
Transport: RTP/AVP;unicast;client_port=39162-39163

Received 87 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 461 Unsupported Transport
CSeq: 3
Server: gortsplib
Session: 3965518070

Failed to SETUP: 461 Unsupported Transport
[151:727][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:461 Unsupported Transport
Failed to PLAY: No RTSP session is currently in progress

[151:727][64] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://192.168.50.186:8555/people/fall_detect error:No RTSP session is currently in progress
```
and this is my html code:
```
<html>
<head>
<script src="adapter.min.js" ></script>
<script src="webrtcstreamer.js" ></script>
<script>        
	var webRtcServer      = null;
	window.onload         = function() { 
		webRtcServer      = new WebRtcStreamer("video","http://192.168.50.186:3456");
		webRtcServer.connect("rtsp://192.168.50.186:8555/people/fall_detect");
	}
	window.onbeforeunload = function() { webRtcServer.disconnect(); }
</script>
</head>
<body> 
	<video id="video" />
</body>
</html>
```

mpromonet, 2023-12-11:
Hi,

It seems strange that ffmpeg identify itself as gortsplib ?

Best Regards,
Michel.

KevinYang123123, 2023-12-12:
hi,thanks for your response.
I tried use vlc to generate the stream.And it works .
Do you mean it can't support simulated rtsp/rtmp stream generated by ffmpeg?
If it can support ffmpeg stream,can you give me one sample?
Regards

yikexiaoshu1, 2024-04-03:
hi,I also encountered this problem, did you solve it?