mpromonet/webrtc-streamer #652: Why does it lag at frame i? Is there a solution，thank you

zhuya1996, 2023-12-20:
Why does it lag at frame i? Is there a solution，thank you

mpromonet, 2023-12-20:
Hi,
I don't understand your problem, but you may submit a PR to fix it.
Best Regards
Michel

zhuya1996, 2023-12-20:
> Hi, I don't understand your problem, but you may submit a PR to fix it. Best Regards Michel

Keyframes are stuck

zhuya1996, 2023-12-20:
> 您好， 我不明白您的问题，但您可以提交 PR 来解决它。 最好的问候 米歇尔

Once the keyframe is reached, it freezes for a while and then returns to normal.

mpromonet, 2023-12-22:
Hi,

This may be a problem of time synchronization, before first RTCP message, stream timestamp is estimated from receiver clock, when RTCP is received, stream timestamp may jump forward or backward.

Best Regards,
Michel.

zhuya1996, 2023-12-26:
> 你好，
> 
> 这可能是时间同步的问题，在第一个RTCP消息之前，流时间戳是从接收器时钟估计的，当接收到RTCP时，流时间戳可能向前或向后跳跃。
> 
> 最好的问候， 米歇尔。

> 你好，
> 
> 这可能是时间同步的问题，在第一个RTCP消息之前，流定时器是从接收器时钟估计的，当接收到RTCP时，流定时器可能向前或光纤跳转。
> 
> 最好的问候， 米歇尔。

Where can I set up the webrtc frame cache? I found that it gets stuck as soon as the key frame is reached. I think it is because the i frame is relatively large and arrives slowly.

zhuya1996, 2024-01-10:
[WeChat_20240110173451.zip](https://github.com/mpromonet/webrtc-streamer/files/13885751/WeChat_20240110173451.zip)
Frame i will get stuck for a while.   thansk

closed 2024-07-18 by zhuya1996