mpromonet/webrtc-streamer #653: "v4l2-ctl" lists devices different to WebRTC interface devices

muhanad40, 2023-12-22:
**Some context**
I have two HDMI capture cards that when connected, they show up under the `/dev` directory. There are two things I'm doing:
- I use Python opencv to connect to these devices like this to do a few things like take screenshots:
  ```python
   import cv2
   cv2.VideoCapture(1, apiPreference=cv2.CAP_V4L2)
   ```
- Use WebRTCStreamer to view the video output of these capture cards

**Describe the bug**
The problem is that when I list the devices using `v4l2-ctl --list-devices`, I get a different list to what shows in the WebRTCStreamer user interface.

```
$ v4l2-ctl --list-devices
USB3.0 HD VIDEO: USB3.0 HD VIDE (usb-0000:00:14.0-3):
	/dev/video0
	/dev/video1
	/dev/media0

USB Video: USB Video (usb-0000:00:14.0-4):
	/dev/video2
	/dev/video3
	/dev/media1
```

When I launch the web interface, I see `video0` and `video1` which is incorrect:
[image]

**Things I tried**
I tried connecting to `video0` and `video1` via opencv, but `video1` didn't work. `video2` worked which makes sense because that's what's also listed in the cli output above.

I tried using the URL in the webrtcstreamer to load `/dev/video2`, but that didn't work:
`http://localhost:8000/?video=videocap://dev/video2`

I tried using the `WebRTCStreamer` JavaScript class to connect to `/dev/video2`, but still didn't work.

Not sure what else to try. I need the webrtcstreamer to be able to connect to actual /dev/devices by path or even index

**To Reproduce**
Steps to reproduce the behavior:
1. Connect multiple capture cards or external cameras
2. Run the webrtcstreamer and check the list of devices and their numbers
3. Open your command line and run `v4l2-ctl --list-devices`

**Expected behavior**
The webrtcstreamer should list the correct device numbers

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser Chrome
 - Version `120.0.6099.71`

mpromonet, 2023-12-23:
Hi @muhanad40 

Please note v4l2:// urls are for H264 encoded stream, then it needs -o argument and H264 support from the v4l2 device. When enabled it should be discovered by /api/getMedialist.    
videocap url used the iteration of video device detected with GetDeviceName see https://github.com/mpromonet/webrtc-streamer/blob/master/inc/CapturerFactory.h#L90 and openned in https://github.com/mpromonet/webrtc-streamer/blob/master/inc/VcmCapturer.h#L81.

Best Regards,
Michel.

muhanad40, 2023-12-23:
Hi @mpromonet, thank you for your quick reply. I don't think the streams are H264 encoded. I just plug the devices in and they show up in the `/dev` directory. How can I connect to a device's webrtc stream using its device path? Is this even possible?

The reason why I ask is because I need a way that's predictable to connect to the webrtc stream. I have a web application that knows a device's /dev path so it can only use that to connect to its stream.

How can I do this?

muhanad40, 2023-12-23:
FYI, I can connect to a device using opencv using the device's /dev path `videocap://dev/video2` and able to grab a screenshot. So I know that `videocap://dev/video*` pattern works. Not sure how I can do this with this.

muhanad40, 2023-12-23:
Another reason why I want to rely on the real /dev path is that it's predictable. Currently the devices are enumerated so I can't predict which device will be which index number. I plan to connect many more devices and it's very important that I use something fixed and not dynamic.

mpromonet, 2023-12-23:
Hi,
You may use device by path. In order to do this, you need to modify the code. Feel free to submit a PR.
Best Regards
Michel

muhanad40, 2023-12-23:
Hi, I wish I can submit a PR, but I'm not a C++ developer. Thanks anyways :)

closed 2023-12-23 by muhanad40