mpromonet/webrtc-streamer #655: 错误：error while loading shared libraries: libXcomposite.so.1: 求助怎么解决？

v5star, 2023-12-28:
我在centos7上运行  ./webrtc-streamer -o 时，报错下面错误，请问大家有遇到吗？怎么解决这个问题。
./webrtc-streamer: error while loading shared libraries: libXcomposite.so.1: cannot open shared object file: No such file or directory

mpromonet, 2023-12-30:
Hi @v5star 

You probably need to install missing libraries. Notice that build is done on ubuntu, using centos may not work. 
On centos you may use snap or docker.

Best Regards,
Michel.

v5star, 2024-01-02:
> Hi @v5star
> 
> You probably need to install missing libraries. Notice that build is done on ubuntu, using centos may not work. On centos you may use snap or docker.
> 
> Best Regards, Michel.
Thank you for your help. I have installed a new Docker container in my CentOS 7.9.2009 x86_64 environment, and it can run properly. I can see the logs when making requests, but there is no video displayed on the webpage. When I refresh the page (the page is from https://github.com/mpromonet/webrtc-streamer-htmlwebrtcstreamer.html), it initially shows a spinning loading icon, and after a few seconds, the screen turns black with no video image. I have attached some screenshots during runtime. Could you please help me identify any misconfigurations and provide a solution? thanks (I have tested the same code on my local Windows platform, and it displays correctly, so I suspect the issue is related to the webrtc-streamer service).
Attachment：
[image]
[image]
[image]
[image]
[image]

mpromonet, 2024-01-02:
Hi,

You should try starting docker with `docker run --net=host`.

Best Regards,
Michel.

v5star, 2024-01-03:
I'm sorry, I still haven't succeeded in displaying the video. I tried adding a container using the image method as shown in the attached picture, as well as trying various command line methods, but it still doesn't work. Can you please point out where my problem might be? Thank you very much.
[image]

command line methods：
docker run -p 8000:8000 -it mpromonet/webrtc-streamer
docker run --name=webrtc --net=host -p 8000:8000 -it mpromonet/webrtc-streamer
docker run --name webrtc --net host -p 8000:8000 -it mpromonet/webrtc-streamer

My container IP is 172.17.0.0/16, and the host IP is 192.168.93.100. The content of config.json is as follows：
{
    "urls":{
	    "10楼电梯口西": {"video": "rtsp://admin:hik12345@192.168.1.88:554/Streaming/Channels/101"},
        "10楼电梯口东" : { "video": "rtsp://admin:hik12345@192.168.1.89:554/Streaming/Channels/101"},
        "10楼北步梯前室": {"video": "rtsp://admin:hik12345@192.168.1.87:554/Streaming/Channels/101"},
        "11楼电梯口东": {"video": "rtsp://admin:hik12345@192.168.1.70:554/Streaming/Channels/101"},
        "11楼电梯口西": {"video": "rtsp://admin:hik12345@192.168.1.69:554/Streaming/Channels/101"},
        "11楼北步梯前室": {"video": "rtsp://admin:hik12345@192.168.1.68:554/Streaming/Channels/101"},
        "12楼电梯口": { "video": "rtsp://admin:hik12345@192.168.1.53:554/Streaming/Channels/101"},
        "12楼北步梯前室": { "video": "rtsp://admin:hik12345@192.168.1.52:554/Streaming/Channels/101"},
        "13楼电梯口东": {"video": "rtsp://admin:hik12345@192.168.1.37:554/Streaming/Channels/101"},
        "13楼北步梯前室": {"video": "rtsp://admin:hik12345@192.168.1.35:554/Streaming/Channels/101"},
        "14楼电梯口东": {"video": "rtsp://admin:hik12345@192.168.1.19:554/Streaming/Channels/101"},
        "14楼电梯口西": {"video": "rtsp://admin:hik12345@192.168.1.18:554/Streaming/Channels/101"},
        "14楼北步梯前室": {"video": "rtsp://admin:hik12345@192.168.1.17:554/Streaming/Channels/101"}
    }
}
Result：
[image]

The command line above works fine for me. Can you please provide an example of how you're using it? I've checked the parameters in the help documentation and they seem to be appropriate for my use case, but I'm still unable to display the video properly.

mpromonet, 2024-01-03:
Hi,
Setting --net=host, the docker IP should be the same than host.
Another could be Google stun server but this don't mind on LAN.
Best Regards
Michel

closed 2024-01-20 by mpromonet