mpromonet/webrtc-streamer #656: webrtc-streamer 播放rtsp视频流时没有音频，望新增

bigsall, 2024-01-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mpromonet, 2024-01-04:
Hi @bigsall 

Did you add an audio rtsp url ?

Best Regards,
Michel.

richard202205, 2024-03-25:
I handled the video stream with webrtc-streamer and succeeded in playing stream on Chrome using angular framework. But there's no audio in this video. I may have the same preoblem with this guy.

richard202205, 2024-03-25:
[image]
This is my code. I added an audio rasp url but there's still no audio in. the video. @mpromonet

zxvgf, 2024-09-04:
Have you solved this audio issue？