mpromonet/webrtc-streamer #658: Can I set the interval time for i-frames during encoding？ thansk

zhuya1996, 2024-01-11:
Can I set the interval time for i-frames during encoding?My desktop resolution may cause lag when others view it

mpromonet, 2024-01-11:
Hi,

You could try to set `cricket::VideoCodec::H264()->keyFrameInterval` given to `webrtc::VideoEncoder::InitEncode`

Best Regards,
Michel.
Best Regards,
Michel.

zhuya1996, 2024-01-12:
> Hi,
> 
> You could try to set `cricket::VideoCodec::H264()->keyFrameInterval` given to `webrtc::VideoEncoder::InitEncode`
> 
> Best Regards, Michel.

May I ask how to set up H264 decoding? I think the default VP8 is currently used, right？

mpromonet, 2024-01-12:
Hi,

codec is negociate between codec supported by client and supported by server. It could be VP8 or H264 or VP9. You could use setCodecPreferences to set preferences.

Best Regards,
Michel.

zhuya1996, 2024-01-15:
> Hi,
> 
> codec is negociate between codec supported by client and supported by server. It could be VP8 or H264 or VP9. You could use setCodecPreferences to set preferences.
> 
> Best Regards, Michel.

In which code is setCodecPreferences set? Can you provide more details? Also, I found that when sharing the screen or reading rtsp, if it is local opencv playback, there will be no lag. If it is web playback, it will lag. Do you know why this is?

zhuya1996, 2024-01-17:
> Hi,
> 
> codec is negociate between codec supported by client and supported by server. It could be VP8 or H264 or VP9. You could use setCodecPreferences to set preferences.
> 
> Best Regards, Michel.

    virtual bool onData(const char *id, unsigned char *buffer, ssize_t size, struct timeval presentationTime)
Is the buffer here live555 data or encoded data?？？

mpromonet, 2024-01-19:
Hi,

This callback receive encoded data.

Best Regards,
Michel.

zhuya1996, 2024-01-20:
> 你好，
> 
> 该回调接收编码数据。
> 
> 最好的问候， 米歇尔。

Hello, why do I lose frames when I play RTSP during the day and a lot at night? How can I make RTSP playback smooth during the day?