mpromonet/webrtc-streamer #661: Vulnerabilities scan result in BDBA（Black Duck Binary Analysi）

manupstairs, 2024-01-22:
Thank you for your great webrtc-streamer!
I need to scan vulnerabilities in BDBA when I use webrtc-streamer in my project.
The problem is that there are some third-party libraries can't be detected in BDBA automatically.
Is there any document or code or link which can help me to indicate these third-party libraries version info?
For example, webrtc-streamer reference FFmpeg, but I can't find FFmpeg version info, the BDBA will report all historical vulnerabilities. 
[image]

mpromonet, 2024-01-22:
Hi,
These third parties are WebRTC ones, you could remove some of them build parameters, especially setting is_chrome_branded=false  https://github.com/mpromonet/webrtc-streamer/blob/master/CMakeLists.txt#L63C19-L63C36 
Best Regards,
Michel.

manupstairs, 2024-05-07:
Hi , according to your guidance, I find something I need at here:
https://chromium.googlesource.com/chromium/third_party/ffmpeg/
Share the infomation with others.
Many thanks, @mpromonet!

closed 2024-05-07 by manupstairs