mpromonet/webrtc-streamer #664: Failed to DESCRIBE: liveMedia0

gOODiDEA2002, 2024-02-06:
**Describe the bug**
docker can't render rtsp video, Failed to DESCRIBE: liveMedia0

**To Reproduce**
1、config.json
---
{
    "urls":{       
        "cam01" : { "video": "rtsp://Admin:A1234567@192.168.200.5:554/Streaming/Channels/101"}
    }
}

2、docker
---
docker run -p 8000:8000 -v $PWD/config.json:/app/config.json mpromonet/webrtc-streamer

3、http://127.0.0.1:8000/
The video playback window remains black

**Expected behavior**
The video window displays the video normally

**Screenshots**
[image]

mpromonet, 2024-02-06:
Hi,

You may get more information increasing log level.

Best Regards,
Michel.

gOODiDEA2002, 2024-02-07:
vlc displays the rtsp video normally

[image]

mpromonet, 2024-02-07:
Hi,

Increasing the log level may help you.

Best Regards,
Michel.

gOODiDEA2002, 2024-02-08:
how to modify the log level?

XiezhDev, 2024-05-30:
> how to modify the log level?

o modify the log level of WebRTC-Streamer, you can use the -v option when running the application. This option controls the verbosity of the logs:

-v for verbose logging
-vv for more verbose logging
-vvv for even more verbose logging

XiezhDev, 2024-05-30:
> how to modify the log level?

To modify the log level of WebRTC-Streamer, you can use the -v option when running the application. This option controls the verbosity of the logs:

-v for verbose logging
-vv for more verbose logging
-vvv for even more verbose logging

lianglll, 2024-06-12:
I have also encountered such problems, how to solve them?

mpromonet, 2024-06-12:
Hi,
Did you try to add `--net host` ?
Best Regards 
Michel

xiaoyf-hub, 2025-01-18:
I encountered the same error, have you resolved it

xiaoyf-hub, 2025-01-18:
I have also encountered such problems, how to solve them?

mpromonet, 2025-01-18:
Hi @xiaoyf-hub 

Which one ?

Best Regards
Michel

xiaoyf-hub, 2025-01-19:
Hi, this is my log, but I'm not sure what the problem is

Failed to DESCRIBE: liveMedia0
[011:117][16228] (rtspvideocapturer.cpp:29): RTSPVideoCapturer:onError url:rtsp://admin:admin123!@172.16.18.151:554/00 error:liveMedia0
Created new TCP socket 1500 for connection
Connecting to 172.16.18.151, port 554 on socket 1500...
...remote connection opened
Sending request: DESCRIBE rtsp://admin:admin123!@172.16.18.151:554/00 RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2024.08.01
Accept: application/sdp

mpromonet, 2025-01-19:
Hi @xiaoyf-hub 

This means sending RTSP DESCRIBE get an error, it might be a problem of encoding with password.

Best Regards,
Michel.