mpromonet/webrtc-streamer #665: Webrtc integrated play RTSP

bingjue123, 2024-02-18:
I am testing Webrtc-streamer to play RTSP video streams;

The RSTP video stream of live video can be played normally, and the video stream address is as follows: "rtsp://admin:tww001218@192.168.105.150:554/dac/realplay/2E691C9D-2E85-4A70-9CBA-482CF5461F9C1/MAIN/TCP?streamform=rtp"

However, the RSTP video stream that plays the video playback cannot be played normally, and the video stream address is as follows: "rtsp://admin:tww001218@192.168.105.150:554/dac/playback/camera/2E691C9D-2E85-4A70-9CBA-482CF5461F9C1/MAIN/TCP?starttime=20240217T050658Z&endtime=20240217T061853Z&name=00000004992000000&size=1064651716&streamform=rtp"
[image]
[image]

bingjue123, 2024-02-18:
In VLC test RTSP address can be played normally

mpromonet, 2024-02-18:
Hi @bingjue123,

Only starttime & ntptime are decoded from rtsp url. Increasing loglevel should help you.
It send a first PLAY, then a PAUSE, then a 2nd PLAY, I don't really understand why not just send PLAY with starttime...

Best Regards,
Michel.

marcelimati, 2024-03-22:
try changing & to ? in url, it should work
Result: rtsp://admin:tww001218@192.168.105.150:554/dac/playback/camera/2E691C9D-2E85-4A70-9CBA-482CF5461F9C1/MAIN/TCP?starttime=20240217T050658Z?endtime=20240217T061853Z?name=00000004992000000?size=1064651716?streamform=rtp

closed 2025-05-17 by mpromonet