mpromonet/webrtc-streamer #668: 在windows环境下（win10，win11）后台运行webrtc-streamer.exe

zhwl123, 2024-03-19:
目前是在windows环境下（win10，win11）运行webrtc-streamer.exe，但是会弹出一个黑窗口，
[image]

这个给客户使用的话显然是不合理的，因此需要在后台运行改程序，这样子就能保证良好的用户感知。
请求大佬指点，如何解决这个问题，万分感谢

yangjieshao, 2024-03-19:
首先 这玩意是让你运行在服务器上的，服务器上一堆程序是命令行的，用户就不用去关心也不会去看服务器上运行的东西
其次 你可以使用 alwaysup 做进程守护和开机自动，alwaysup 启动的程序默认运行在  session0 也就是后台会话

zhwl123, 2024-03-19:
邮件已收到，将尽快回复您！

yangjieshao, 2024-03-19:
```
var procInfo = new ProcessStartInfo(exeFullPath, Arguments)
  {
      WorkingDirectory = new System.IO.FileInfo(exeFullPath).DirectoryName,
      UseShellExecute = false,
      CreateNoWindow = true,
      RedirectStandardInput = true,
      RedirectStandardOutput = true,
      RedirectStandardError = true
  };
Process.Start(procInfo);
```

你也可以写代码 这样启动外部命令行程序

liusuyi2021, 2024-03-19:
> ```
> var procInfo = new ProcessStartInfo(exeFullPath, Arguments)
>   {
>       WorkingDirectory = new System.IO.FileInfo(exeFullPath).DirectoryName,
>       UseShellExecute = false,
>       CreateNoWindow = true,
>       RedirectStandardInput = true,
>       RedirectStandardOutput = true,
>       RedirectStandardError = true
>   };
> Process.Start(procInfo);
> ```
> 
> 你也可以编写代码这样启动外部命令行程序
你还真是手把手教啊，这种选手就不适合玩开源流媒体了

zhuya1996, 2024-03-19:
兄弟，你人才，翻都不翻译  哈哈

zhwl123, 2024-03-19:
> ```
> var procInfo = new ProcessStartInfo(exeFullPath, Arguments)
>   {
>       WorkingDirectory = new System.IO.FileInfo(exeFullPath).DirectoryName,
>       UseShellExecute = false,
>       CreateNoWindow = [...]

我们的场景比较特殊，服务端就要在客户机上，然后浏览器中是客户端调用

zhwl123, 2024-03-19:
> 兄弟，你人才，翻都不翻译 哈哈

要让中文走向世界撒

zhwl123, 2024-03-19:
> ```
> var procInfo = new ProcessStartInfo(exeFullPath, Arguments)
>   {
>       WorkingDirectory = new System.IO.FileInfo(exeFullPath).DirectoryName,
>       UseShellExecute = false,
>       CreateNoWindow = [...]

我们的场景比较特殊，服务端就要在客户机上，然后浏览器中是客户端调用。主要是对视频的响应有要求，不然就用专门的服务器安装了。

yangjieshao, 2024-03-20:
> 我们的场景比较特殊，服务端就要在客户机上，然后浏览器中是客户端调用。主要是对视频的响应有要求，不然就用专门的服务器安装了。

所以 你猜为什么我的回答里有说 使用alwaysup 和自己写代码两种方式？
如果这两种方式都不行，我建议你换工作
其次，你们的场景一点都不特殊，内网部署的服务器也不存在响应慢的问题
甚至通过frp（访问内网部署的网页+后台程序）+tun服务（访问视频流） 远程看视频都没啥延迟，响应非常快

yangjieshao, 2024-03-20:
至于为什么不是sun服务而是tun服务 参考 https://github.com/mpromonet/webrtc-streamer/issues/552

liusuyi2021, 2024-03-20:
杨杰少说的不错，流媒体从来就没有运行在客户端的这种操作，你认为局域网延迟能有多大？现在的人用浏览器的优势之一 就是不用安装直接打开即可访问，你非要部署在客户机无疑是走了回头路；另外，就事论事，让exe后台启动这种基本问题你还在GitHub上问，显得你不是一个善于百度和chatgpt的人，开发这行可能不适合你

closed 2024-03-20 by zhwl123

zhwl123, 2024-08-02:
邮件已收到，将尽快回复您！