mpromonet/webrtc-streamer #669: Run on Raspberry Pi 4 with Raspberry Pi OS Bookworm

ZakEspley, 2024-03-20:
Hi,

I am not particularly new to using github or a raspberry pi, but I can't see to figure out how to actually run this software. Do I need to compile it for myself? If so, what are the releases for?

I have downloaded the release for the armv764 version as this should match the Raspberry pi 4 running 64 bit Rapberry Pi OS Bookworm.

I extract the file and cd into directory. Then when I try to run 
```
./webrtc-streamer
```
I get
```
bash: ./webrtc-streamer: cannot execute: required file not found
```
as a response.

The webrtc-streamer file is there when do an `ls` command so I am thinking it depends on some sort of dependency or library file that I don't have. 

What things will I need to install in addition to downloading a release in order to get this great software to work? Additionally, maybe there could be some sort of installation for Dummies section in the README?

Thanks,
Zak

ZakEspley, 2024-03-20:
I see now my mistake. I guess it's my first time using Bookworm and the modern 64 bit version, but when I downloaded the arm64 release and used that everything worked.

I do have another question though. I am using an older model OV5647 bases raspberry pi CSI camera. It does not have native H264 encoding, only yuyv.

Can I have camera streamer collect the data from that with v4l2? If so, what is the command I need to run to make that happen? I tried setting the URL with the -u flag to v4l2:///dev/video0 but it says it can't access the camera. Is my command wrong or is there another way for me to get this to work? Do I need to use ffmpeg to convert to h264 first? Or maybe rstp?

Any help would be great.

mpromonet, 2024-03-20:
Hi,

V4l2 capture works only with h264, in order to forward compressed frame. If your camera produce uncompressed format, just use normal capture.
Best Regards
Michel

ZakEspley, 2024-03-21:
So does that mean that I set the url to videocap:///dev/video0 ?

mpromonet, 2024-03-22:
Hi,

I guess probably videocap://0, but you could get it using the api /api/getVideoDeviceList

Best Regards,
Michel.

closed 2024-05-18 by mpromonet