mpromonet/webrtc-streamer #670: Unknown problem with arm version

wangTaoTaox, 2024-03-22:
No screen, also do not load, please see the following log

Version:v0.6.1/Linux-arm64 civetweb@v1.13 webrtc@18410aa438-dirty live555helper@897450c
nullLogger level:3
xcb_connection_has_error() returned true
[000:000][1070787] (adm_helpers.cc:74): Failed to query stereo recording.
[000:000][1070787] (audio_device_generic.cc:18): BuiltInAECIsAvailable: Not supported on this platform
[000:000][1070787] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:000][1070787] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform
HTTP Listen at 0.0.0.0:8000
[033:153][1070787] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.7381466261833769
[033:155][1070787] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.7381466261833769
[033:156][1070787] (webrtc_video_engine.cc:3363): Absent receive stream; ignoring clearing encoded frame sink for ssrc 0
[033:156][1070787] (PeerConnectionManager.h:214): virtual void PeerConnectionManager::PeerConnectionObserver::OnAddStream(rtc::scoped_refptr<webrtc::MediaStreamInterface>) nb video tracks:1
[033:156][1070796] (CapturerFactory.h:246): audiourl: rtsp://admin:HK123456@192.168.1.64:554/Streaming/Channels/2 idx_audioDevice:-1/1
[033:156][1070796] (PeerConnectionManager.cpp:1150): Cannot create capturer video: rtsp://admin:HK123456@192.168.1.64:554/Streaming/Channels/2
[033:156][1070796] (PeerConnectionManager.cpp:1167): Cannot create capturer audio: rtsp://admin:HK123456@192.168.1.64:554/Streaming/Channels/2
[033:156][1070787] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.7381466261833769
[033:158][1070787] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.7381466261833769
[033:158][1070793] (audio_device_pulse_linux.cc:1861): Can't query latency
[033:278][1070794] (stun_port.cc:308): Port[500062a0:0:1:0:local:Net[docker0:172.17.0.x/16:Unknown:id=1]]: UDP send of 96 bytes failed with error 101
[033:338][1070787] (PeerConnectionManager.h:144): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ServerDataChannel state:open
[033:341][1070787] (PeerConnectionManager.h:225): virtual void PeerConnectionManager::PeerConnectionObserver::OnDataChannel(rtc::scoped_refptr<webrtc::DataChannelInterface>)
[033:341][1070787] (PeerConnectionManager.h:144): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ClientDataChannel state:open
[033:341][1070787] (PeerConnectionManager.h:151): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ClientDataChannel msg:local channel openned
[033:342][1070787] (PeerConnectionManager.h:151): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ServerDataChannel msg:remote channel openned
[033:470][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101
[033:518][1070794] (stun_port.cc:308): Port[500062a0:0:1:0:local:Net[docker0:172.17.0.x/16:Unknown:id=1]]: UDP send of 96 bytes failed with error 101
[033:566][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101
[033:614][1070794] (stun_port.cc:308): Port[500062a0:0:1:0:local:Net[docker0:172.17.0.x/16:Unknown:id=1]]: UDP send of 96 bytes failed with error 101
[033:662][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101
[034:210][1070794] (stun_port.cc:308): Port[500062a0:0:1:0:local:Net[docker0:172.17.0.x/16:Unknown:id=1]]: UDP send of 96 bytes failed with error 101
[035:211][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101
[036:212][1070794] (stun_port.cc:308): Port[500062a0:0:1:0:local:Net[docker0:172.17.0.x/16:Unknown:id=1]]: UDP send of 96 bytes failed with error 101
[036:713][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101
[037:713][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101
[039:214][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101
[040:215][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101
[041:717][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101
[042:717][1070794] (stun_port.cc:308): Port[5002bad0:0:1:0:local:Net[enp11s0f0:192.168.1.x/24:Unknown:id=2]]: UDP send of 96 bytes failed with error 101

What's the reason

wangTaoTaox, 2024-03-22:
Two servers, same code and plugin, one can play

wangTaoTaox, 2024-03-23:
@mpromonet  
Because GLIBC is version 2.28, we can use versions 0.52, 0.61, 0.62, 0.63, and all the above problems occur. In this case, we can only upgrade GLIBC?

Best Regards,WangTao

mpromonet, 2024-03-23:
Hi @wangTaoTaox 

You may use docker ?

Best Regards,
Michel.

wangTaoTaox, 2024-03-23:
Hi@mpromonet 

docker was also unavailable.

Best Regards,

WangTao

mpromonet, 2024-03-23:
Hi,

/usr/include/asm-generic/errno.h:#define        ENETUNREACH     101     /* Network is unreachable */

Probably some problem with the network config ?

Best Regards,
Michel.

wangTaoTaox, 2024-03-23:
Hi@mpromonet 

The Internet can be spelled

Best Regards,
WangTao.

mpromonet, 2024-03-23:
Hi,

We see 2 networks on your log:
- docker0:172.17.0.x/16
- enp11s0f0:192.168.1.x/24

Then tshark or strace may helps, maybe wrong default route....

Best Regards,
Michel.

closed 2024-08-04 by mpromonet