mpromonet/webrtc-streamer #674: mac specifies that rtsp cannot get video, but windows can

zhou1591, 2024-03-27:
**Describe the bug**

Hi everyone, please ask a question, I have a camera on the Intranet, and then I use webrtc-streamer to link my camera under windows and mac systems respectively, using the official demo. Then I was fine in Windows, but I got an error on mac

```
connect error: SyntaxError: Failed to construct 'RTCPeerConnection': ICE server parse failed
```

He is in WebRtcStreamer. Prototype. The enclosing createPeerConnection PC = new RTCPeerConnection (enclosing pcConfig) error here

On the same camera device, my command to start is the same

```
./webrtc streamer - 8000 - a - N10 - o - H S RTSP://192.168.100.37/live/main_stream
```

Then I printed the source of this.pc, the data returned from the /api/getIceServers api

I found that their returns were different

windows returns the result as

```
{
  "iceServers" :
    [
      {
        "urls" :
          [
            "Stun: 127.0.0.1:3478"
          ]
      }
    ],
  "iceTransportPolicy" : "all"
}

```

The return result for mac is

```
{
  "iceServers" :
    [
      {
        "urls" :
          [
            Stun: "RTSP: / / 192.168.100.47 / live/main_stream"
          ]
      }
    ],
  "iceTransportPolicy" : "all"
}

```

I think it is "stun: RTSP: / / 192.168.100.47 / live/main_stream" caused by the error

What should I do to make my mac receive the video stream normally?

Thank you all

mpromonet, 2024-03-27:
Hi,
It seems you made confusion between ICE/stun and video URL 
Best Regards
Michel

zhou1591, 2024-03-28:
> Hi, It seems you made confusion between ICE/stun and video URL Best Regards Michel

Hello, I don't understand what you mean. I use the same command in both win and mac

```
./webrtc streamer - 8000 - a - N10 - o - H S RTSP://192.168.100.37/live/main_stream
```

The result of this return is what /api/getIceServers returns to me
I also feel that his return result is a little strange, I would like to ask what I did wrong? How can I get this api to return a correct result? I'm using the official demo

Thank you for your answer. Please help me. Thank you

zhou1591, 2024-03-28:
> Hi, It seems you made confusion between ICE/stun and video URL Best Regards Michel

I downloaded it from here

[webrtc-streamer-v0.8.5-Darwin-x86_64-Release.tar.gz](https://github.com/mpromonet/webrtc-streamer/releases/download/v0.8.5/webrtc-streamer-v0.8.5-Darwin-x86_64-Release.tar.gz)

mpromonet, 2024-03-28:
Hi,
The arguments contains wrong space characters between minus and options.
Best Regards
Michel

zhou1591, 2024-03-28:
> Hi, The arguments contains wrong space characters between minus and options. Arguments parsing is a bit different on windows. Best Regards Michel

[image]

Yes such parameters may have some problems with git typesetting

zhou1591, 2024-03-28:
> Hi, The arguments contains wrong space characters between minus and options. Arguments parsing is a bit different on windows. Best Regards Michel

Please look at this picture. Thank you

zhou1591, 2024-03-28:
> Hi, The arguments contains wrong space characters between minus and options. Arguments parsing is a bit different on windows. Best Regards Michel

this is mac

[image]

this is win

[image]

mpromonet, 2024-03-28:
Hi,
Your problem is that -S takes rtsp URL as argument value. With a space it should not. Anyway, you can put it as last option, or add default value.
Best Regards
Michel

zhou1591, 2024-03-28:
> Hi, Your problem is that -S takes rtsp URL as argument value. With a space it should not. Anyway, you can put it as last option, or add default value. Best Regards Michel

Ok, thank you. I misunderstood the usage of this parameter, and the problem has been solved. thank you.

closed 2024-03-28 by zhou1591