mpromonet/webrtc-streamer #676: How to add a tip?

nianhuayixiao, 2024-04-02:
hi,

If a certain Hikvision camera does not start or is not connected to the internet, which code will prompt on the front-end page?

v0.8.5

CHN-STUDENT, 2024-06-17:
我用 vue2 + element-ui 写的两个遮罩层 loading 和 loading Error
```
 setPlay(url) {
            // 检查 WebRtcStreamer 是否已加载并创建实例
            if (typeof WebRtcStreamer !== 'undefined') {
                const videoElement = document.getElementById('videoWrapper');
                this.videoPlayerPlugin = new WebRtcStreamer(videoElement, location.protocol + "//172.16.172.24:8000");

                // 连接视频流
                this.videoPlayerPlugin.connect(url);

                // 设置计时器，如果10秒后未播放则认为播放失败
                const timer = setTimeout(() => {
                    this.loading = false;
                    this.loadingError = true;
                    videoElement.removeEventListener('playing', onPlaying);
                    // videoElement.setAttribute("controls", true); // 显示控制菜单
                }, 10000);

                // 播放成功的回调
                const onPlaying = () => {
                    clearTimeout(timer);
                    this.loading = false;
                    this.loadingError = false;
                    videoElement.removeAttribute("controls"); // 隐藏控制菜单
                };

                videoElement.addEventListener('playing', onPlaying);

                // 播放失败的回调
                this.videoPlayerPlugin.onError = (error) => {
                    console.error("WebRTC error: ", error);
                    clearTimeout(timer);
                    this.loading = false;
                    this.loadingError = true;
                    videoElement.setAttribute("controls", true); // 显示控制菜单
                };

            } else {
                this.loading = false;
                this.loadingError = true;
            }
        }
```

CHN-STUDENT, 2024-06-18:
@mpromonet hi,mpromonet could you give me some advice to use in my vue project? our old project only use vue2, i try to use your project instead of RTMP(flash)

closed 2024-11-11 by mpromonet