mpromonet/webrtc-streamer #677: rtsp video stream cannot be played

yikexiaoshu1, 2024-04-03:
Hello, I used VLC to simulate an rtsp video stream, such as rtsp://172.16.50.12:8554/v. I installed webrtc-streamer in window and linux and played the rtsp video stream through webrtcstreamer.html. The webrtc-streamer will report an error. 
Error content: 
Failed to SETUP: 461 Unsupported transport
Failed to SETUP: 461 Unsupported transport
Failed to PLAY: No RTSP session is currently in progress.

How should I solve it? Thank you.

yikexiaoshu1, 2024-04-03:
In VLC test RTSP address can be played normally

mpromonet, 2024-04-03:
Hi,
By default, it use rtp over TCP.
Best Regards
Michel

yikexiaoshu1, 2024-04-03:
Hello, I don’t understand what you mean. Can you explain it in detail?How can I set it up to play rtsp video stream?

mpromonet, 2024-04-03:
Hi @yikexiaoshu1,

The default options are defined in https://github.com/mpromonet/webrtc-streamer-html/blob/f27dc4ca97aab57ce1f85589983f867a197a70cd/webrtcconfig.js, then
`options: "rtptransport=tcp&timeout=60",`

You could overide it adding in url  `/webrtcstreamer.html?video=.....&options=` or `/webrtcstreamer.html?video=.....&options=rtptransport%3Dudp` 

Best Regards,
Michel.

closed 2024-04-10 by yikexiaoshu1