mpromonet/webrtc-streamer #681: How to get audio

yikexiaoshu1, 2024-04-07:
With this code, we like to get video & audio.But after trying, there is still no audio, the audio stream encoding is aac, what should we do?
webRtcServer.connect("rtsp://172.16.12.50:554/stream","rtsp://172.16.12.50:554/stream");

zhuya1996, 2024-04-23:
> 通过这段代码，我们想要获取视频和音频。但是尝试之后，仍然没有音频，音频流编码是aac，我们该怎么办？ webRtcServer.connect("rtsp://172.16.12.50:554/stream","rtsp://172.16.12.50:554/stream");

rtsp set audio G711_U

mpromonet, 2024-04-26:
Hi @zhuya1996,

audio codec support is checked https://github.com/mpromonet/webrtc-streamer/blob/master/inc/liveaudiosource.h#L94, when not supported you should see error "LiveAudioSource::onNewSession not support codec" (see  https://github.com/mpromonet/webrtc-streamer/blob/master/inc/liveaudiosource.h#L102)

Best Regards,
Michel.

closed 2024-12-30 by mpromonet