mpromonet/webrtc-streamer #687: I decided to use datachannel to transmit h265 and then decode it locally. Can you teach me how to use datachannel?

zhuya1996, 2024-04-12:
I decided to use datachannel to transmit h265 and then decode it locally. Can you teach me how to use datachannel?  thanks

mpromonet, 2024-04-12:
Hi,
I feel this approach strange, in such a case why using webrtc, just use a standard websocket.
Best Regards 
Michel

zhuya1996, 2024-04-15:
datachannel has relatively better bandwidth control capabilities, better latency, and better data transmission security.

zhuya1996, 2024-04-15:
> 嗨， 我觉得这种方法很奇怪，在这种情况下为什么使用 webrtc，只使用标准的 websocket。 您可以查看使用 H265 的https://github.com/mpromonet/rtsp2web 。 最好的问候 米歇尔

datachannel has relatively better bandwidth control capabilities, better latency, and better data transmission security.

zhuya1996, 2024-04-17:
> Hi, I feel this approach strange, in such a case why using webrtc, just use a standard websocket. You may look to https://github.com/mpromonet/rtsp2web that is working with H265. Best Regards Michel

Do you know that the datachannel project has GCC congestion algorithm?

closed 2024-04-23 by zhuya1996

mpromonet, 2024-05-08:
Hi @zhuya1996 

You may look to https://www.w3.org/TR/webtransport/

Best Regards,
Michel.