mpromonet/webrtc-streamer #690: After a period of time, frames are dropped and reconnection is impossible

zhou1591, 2024-04-24:
## Hello everyone, please ask a few questions, I hope you can help me

### I am now using the win10 above the official website example, my camera push rtsp video stream, the code is H264, the size of the video is 1280*720, the wifi bandwidth is about 20M, after I use the example to start sometimes there will be frame loss

[image]

### Can he disconnect the rtsp video stream himself and reconnect it when he appears the same message as in the above figure, so it seems that he will keep dropping as long as there is a dropped frame error

### Then there is another problem, after I close the exe inside the official sample program, I open it again, sometimes the backend service api will report an error directly in the js

```
WebRtcStreamer.prototype.onError
```

mpromonet, 2024-04-25:
Hi,

It is usually a good practice to ask only one question in a thread. Otherwise it becomes quickly messy.
Reading quickly I don't know what is rtsp_demo.c, it doesnot seems to come from this project.
However when there is some frame drops, it may be possible to take some decisions.

Best Regards 
Michel

zhou1591, 2024-05-08:
> Hi,
> 
> It is usually a good practice to ask only one question in a thread. Otherwise it becomes quickly messy. Reading quickly I don't know what is rtsp_demo.c, it doesnot seems to come from this project. However when there is some frame drops, it may be possible to take some decisions. Nowdays connection is closed (and then restarted) when no data is received for an amount of time.
> 
> Best Regards Michel

Is there any way I can catch a dropped frame error in the exe and re-establish the link without closing the exe?

zhou1591, 2024-05-20:
[image]

> Hi,
>
> It is usually a good practice to ask only one question in a thread. Otherwise it becomes quickly messy. Reading quickly I don't know what is rtsp_demo.c, it doesnot seems to come from this [...]

Sometimes the program itself shows this up

closed 2024-06-06 by zhou1591