mpromonet/webrtc-streamer #697: Which version of webrtc source code match with webrtc-streamer?

howardhuang666, 2024-05-20:
I used newest webrtc and 2023-12-06T04:11:35, but neither can match with it
[image]

mpromonet, 2024-05-20:
Hi,
Latest, it may need adaptation time to Time.
Best Regards 
Michel

howardhuang666, 2024-05-22:
> Hi, Latest, it may need adaptation time to Time. 2023 seems quite old.... latest one today is fa4128eedbe427eecd28988f2585753645514296, and seems to build. Best Regards Michel

[image]

Hi, Michel
Maybe I was really bad in git,
I couldn't find version fa4128eedbe427eecd28988f2585753645514296
Can you help me a little bit?

mpromonet, 2024-05-22:
Hi,

git pull origin master may helps you.

Best Regards 
Michel

closed 2024-06-04 by mpromonet