mpromonet/webrtc-streamer #699: Whether the X86 architecture of mac computers is supported

zhou1591, 2024-06-06:
We have some users whose computers may be old mac computers, their cpu is inter chip, is x86 architecture, does this project support?

mpromonet, 2024-06-08:
Hi @zhou1591,

The macos support was kindly add by @prife in https://github.com/mpromonet/webrtc-streamer/pull/455. I don't have any usage of macos, I just maintain the pipeline. 
Recently as it was broken, I update build image to github action macos-latest that seems arm64.
There is no more pipeline for macos on x86, I guess you can try to build it.

Best Regards,
Michel.

closed 2024-06-13 by zhou1591

xiaoxiaohuayu, 2025-02-07:
Has anyone built a package for mac x86? I'm a front-end engineer and not very familiar with this compilation. Can someone help me?