mpromonet/webrtc-streamer #706: Run webcam USB 0 in window

duy1851999, 2024-07-10:
Hi sir, How to run webrtc-streamer by webcam USB 0 in window?

mpromonet, 2024-07-10:
Hi @duy1851999 

USB camera should be detected on windows, what is your question ?

Best Regards 
Michel

duy1851999, 2024-07-10:
I run docker run --device=/dev/video0 -p 8000:8000 -it mpromonet/webrtc-streamer in ubuntu, it run ok. But when I change to window and run like that: docker run --device=/dev/video0 -p 8000:8000 -it mpromonet/webrtc-streamer or docker run --device=0 -p 8000:8000 -it mpromonet/webrtc-streamer . It has the error:
docker: Error response from daemon: error gathering device information while adding custom device "/dev/video0": no such file or directory. The question is How to run the webrtc on window(not linux)?

duy1851999, 2024-07-10:
I think when I run docker on window, the docker is not detect USB camera, but it ok with rtsp camera

mpromonet, 2024-07-10:
Hi @duy1851999 

You can access the camera running webrtc-streamer as a windows application. 
If you link to bind usb device to wsl you may look to https://learn.microsoft.com/en-us/windows/wsl/connect-usb.

Best Regards,
Michel.