mpromonet/webrtc-streamer #708: RSTP Video display screen

GuanTu, 2024-07-25:
**Describe the bug**
Use /webrtcstreamer.html to access the RSTP video address. The video page of the browser will display a flower screen. See the attached pictures and logs for more information

**Screenshots**
[image]

**Logs**
[192.168.16.103-GIS_2024-07-25_9_10_35.log](https://github.com/user-attachments/files/16369028/192.168.16.103-GIS_2024-07-25_9_10_35.log)

**Desktop (please complete the following information):**
 - OS:  Windows 10
 - Browser: Chrome
 - Version: 126.0.6478.183

PS: English is not my native language. Please forgive me if there are any mistakes through translation.  :)

GuanTu, 2024-07-25:
Playing RSTP addresses using vlc is ok

wzk1583456343, 2024-10-29:
您好请问您解决这个问题了嘛，我也碰到了一样的问题

mpromonet, 2024-11-01:
Hi @GuanTu 

In your log, we see:
```
[74879:813][315] (video_stream_encoder.cc:1517): Same/old NTP timestamp (1721869940461 <= 1721869940461) for incoming frame. Dropping.
[74879:813][315] (frame_cadence_adapter.cc:1001): Incoming frame timestamp is not monotonically increasing current: 1721869940461000 last: 1721869940461000
```
This might be origin of your problem. 

You can try to remove the '-o' to not forward encoded stream, and then transcode it.

Best Regards,
Michel

P.S. English is not my native language. Please forgive me if there are any mistakes through translation. :)

GuanTu, 2024-11-04:
> Hello, have you solved this problem? I also encountered the same problem.

Sorry, I did not find a solution to this problem, you can refer to mpromonet's reply

GuanTu, 2024-11-04:
> Hi @GuanTu
>
> In your log, we see:
>
> ```
> [74879:813][315] (video_stream_encoder.cc:1517): Same/old NTP timestamp (1721869940461 <= 1721869940461) for incoming frame. Dropping.
> [74879:813][315] [...]

Thank you for your reply. If there is a chance in the future, I will try it

wzk1583456343, 2024-11-04:
> Hi  你好@GuanTu
> 
> In your log, we see:在您的日志中，我们看到：
> 
> ```
> [74879:813][315] (video_stream_encoder.cc:1517): Same/old NTP timestamp (1721869940461 <= 1721869940461) for incoming frame. Dropping.
> [74879:813][315] (frame_cadence_adapter.cc:1001): Incoming frame timestamp is not monotonically increasing current: 1721869940461000 last: 1721869940461000
> ```
> 
> This might be origin of your problem.这可能是您问题的根源。
> 
> You can try to remove the '-o' to not forward encoded stream, and then transcode it.您可以尝试删除“-o”以不转发编码流，然后对其进行转码。
> 
> Best Regards, 此致， Michel 米歇尔
> 
> P.S. English is not my native language. Please forgive me if there are any mistakes through translation. :)PS 英语不是我的母语。翻译如有错误，敬请谅解。 :)

After I deleted the -o parameter, half of the video was no longer mosaic, but I could clearly see a dividing line between the upper and lower parts of the video. This problem would be even more obvious if the video is in motion. Is there any solution for this? Thank you very much.