mpromonet/webrtc-streamer #709: I use version 0.8.3 to play rtsp. If I don't play the audio, the video is normal. If I also play the audio, the video will be delayed after a period of time. How can I modify this?

zhuya1996, 2024-07-30:
I use version 0.8.3 to play rtsp. If I don't play the audio, the video is normal. If I also play the audio, the video will be delayed after a period of time. How can I modify this?  thanks

closed 2024-10-14 by zhuya1996

zhuya1996, 2024-11-14:
你先用vlc 播放器看看 有没有声音，然后切换到0.83的版本 ，没解决再说，解决了记得发个红包给我， 这个项目 我硬解码编码 我都实现了 

莫失莫忘0212
***@***.***

 
@zhuya1996   怎么播放音频呀，
我这样设置了 声音没放出来

yangjieshao, 2024-11-14:
启动命令是？
我用的启动命令是  -o -s-

zhuya1996, 2024-11-14:
-0 是转发啊   不进行编解码 
 

莫失莫忘0212
***@***.***

yangjieshao, 2024-11-14:
所以 不能用  `-o`  命令？
主要是 不用 `-o`  cpu吃不消啊

yangjieshao, 2024-11-15:
0.8.3 使用 `-o` 可以有声音
看样子是 0.8.7 有问题了

@mpromonet 

same command 
```
webrtc-streamer -o -s-
```
same code
```
webRtcServer.connect("rtsp://admin:pwd@192.168.1.95:554/Streaming/Channels/101"
		,"rtsp://admin:pwd@192.168.1.95:554/Streaming/Channels/101");
```

audio ok in v0.8.3
no audio in v0.8.7

zhuya1996, 2024-11-15:
那就是解决了咯&nbsp;&nbsp;&nbsp;有木有红包
&nbsp;

莫失莫忘0212
***@***.***

&nbsp;

------------------&nbsp;原始邮件&nbsp;------------------
发件人: ***@***.***&gt;; 
发送时间: 2024年11月15日(星期五) 上午10:43
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [mpromonet/webrtc-streamer] I use version 0.8.3 to play rtsp. If I don&#39;t play the audio, the video is normal. If I also play the audio, the video will be delayed after a period of time. How can I modify this? (Issue #709)

0.8.3 使用-o可以有声音
看样子是 0.8.7 有问题了

@普罗莫内

相同的命令
webrtc-streamer-o-s-
相同的代码
***@***.******@***.***:554/流媒体/频道/101”);
版本0.8.3中的音频正常
版本0.8.7中没有音频

-
直接回复这封邮件，在GitHub上查看，或取消订阅.
***@***.*** &gt;

yangjieshao, 2024-11-15:
> 那就是解决了咯&nbsp;&nbsp;&nbsp;有木有红包

问题是 怎么发

zhuya1996, 2024-11-15:
+qq  2308193190  这个项目我已经完全弄懂了，有其它问题也可以咨询我 

莫失莫忘0212
***@***.***

  
那就是解决了咯   有木有红包
  
问题是 怎么发