mpromonet/webrtc-streamer #710: The delay of using audio and video is much greater than that of playing video alone. Please tell me why this is happening. I am waiting for your reply. Thank you.

zhuya1996, 2024-08-02:
[image]
[image]

mpromonet, 2024-08-03:
Hi @zhuya1996,

It might be because there is 2 rtsp connections, it may also have some improvement to do in timestamp publish to webrtc sdk...

Best Regards,
Michel.

zhuya1996, 2024-08-04:
> 你好@zhuya1996，
> 
> 这可能是因为有 2 个 rtsp 连接，在将时间戳发布到 webrtc sdk 时可能也需要做一些改进...
> 
> 此致敬礼， 米歇尔。

> Hi @zhuya1996,
> 
> It might be because there is 2 rtsp connections, it may also have some improvement to do in timestamp publish to webrtc sdk...
> 
> Best Regards, Michel.

1. Should I modify the sent timestamp in webrtc-streamer or in the webrtc source code?
2. If I remove the audio and video synchronization, I wonder if it can solve the problem of video delay. What do you think?

closed 2024-10-14 by zhuya1996