mpromonet/webrtc-streamer #713: WebRtcStreamer.disconnect() not release connection, that make max preview limit and camera preview broken

cky-thinker, 2024-08-16:
**Describe the bug**
I use webrtc-streamer to transform HikCamera rtsp to webrtc and play on web.

I found WebRtcStreamer.disconnect() not release connection, that make max preview limit(10) and camera preview broken

The connection num of camera will release only when i restart webrtc-streamer server

**Expected behavior**
WebRtcStreamer.disconnect() should release

mpromonet, 2024-08-16:
Hi @cky-thinker 

Disconnect, close the webrtc stream, backend connection is shared and is released when no connection still use it.

Best Regards 
Michel

cky-thinker, 2024-08-19:
Because connection not close, that will make camera max preview limit(10) and camera preview broken

mpromonet, 2024-08-19:
Hi @cky-thinker 
Then it may have some problem closing the rtsp session. You may look rtsp and rtcp bye.
Best Regards 
Michel

commit 946007b3 (2024-08-19): fix #713
closed 2024-08-19 by mpromonet